rusefi/rusefi #6502: MAX3185x - SPI5 crash

pchmura4, 2024-05-12:
**Describe the bug**
When SPI5 is selected for EGT input, the ECU crashes on `spiStart(driver, &spiConfig);` in _firmware/hw_layer/sensors/max3185x.cpp_ in `spi_txrx` function.

**Hardware**
Board: Proteus

rusefillc, 2024-05-12:
@pchmura4 please provide more info. What does 'crash' mean? If you know function name maybe you can point at specific line?

rusefillc, 2024-05-12:
@pchmura4 https://rusefi.com/online/ is the tool we use so that humans would not have to type "CS => PF6
SCK => PF7
MISO => PF8" expecting other humans to manually take those parameters.

pchmura4, 2024-05-12:
> @pchmura4 please provide more info. What does 'crash' mean? If you know function name maybe you can point at specific line?

No connection to ECU, not detected on USB bus anymore, that's what I meant by crash.
The only way to reflash after selecting SPI5 is to hold PROG while connecting power to enter bootloader.

When the line 234 is executed, the ECU stops responding and crashes: https://github.com/rusefi/rusefi/blob/01db172eaec5719ba538d6a52f02aa28bfab804b/firmware/hw_layer/sensors/max3185x.cpp#L234

> @pchmura4 https://rusefi.com/online/ is the tool we use so that humans would not have to type "CS => PF6 SCK => PF7 MISO => PF8" expecting other humans to manually take those parameters.

This is not in TunerStudio, I added this so humans don't have to manually look on Proteus schematic and check what pins I connected.
Maybe this will be better reference:
https://github.com/rusefi/rusefi/blob/01db172eaec5719ba538d6a52f02aa28bfab804b/firmware/config/boards/proteus/board_configuration.cpp#L137-L140

pchmura4, 2024-05-12:
This can also be helpful
[image]
[image]

dron0gus, 2024-05-12:
Is this a f4 proteus?
Tested on f7 - not reproduced.

dron0gus, 2024-05-12:
No DMA is allocated for SPI5.
```
Thread 2 "MAX3185X" received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 536889992]
chDbgPanic3 (msg=0x80714d8 <__func__.4.lto_priv.8> "spi_lld_start", file=0x8071244 "ChibiOS/os/rt/src/chsys.c", line=201) at ./controllers/core/error_handling.cpp:108
108		__asm volatile("BKPT #0\n");
(gdb) bt
#0  chDbgPanic3 (msg=0x80714d8 <__func__.4.lto_priv.8> "spi_lld_start", file=0x8071244 "ChibiOS/os/rt/src/chsys.c", line=201) at ./controllers/core/error_handling.cpp:108
#1  0x08002a30 in chSysHalt (reason=reason@entry=0x80714d8 <__func__.4.lto_priv.8> "spi_lld_start") at ChibiOS/os/rt/src/chsys.c:201
#2  0x08004c5a in spi_lld_start (spip=spip@entry=0x2000865c <SPID5>) at ChibiOS/os/hal/ports/STM32/LLD/SPIv1/hal_spi_lld.c:406
#3  0x08004cce in spiStart (spip=0x2000865c <SPID5>, config=0x80714d8 <__func__.4.lto_priv.8>) at ChibiOS/os/hal/src/hal_spi.c:99
#4  0x0806b97a in Max3185xRead::spi_txrx (this=this@entry=0x200047a0 <instance>, ch=ch@entry=0, tx=tx@entry=0x20004a20 <instance+640> "\200\001UUUUUUUUUU\240G", 
    rx=rx@entry=0x20004a24 <instance+644> "UUUUUUUU\240G", n=n@entry=2) at ./hw_layer/sensors/max3185x.cpp:234
#5  0x08034784 in Max3185xRead::detect (this=this@entry=0x200047a0 <instance>, ch=ch@entry=0) at ./hw_layer/sensors/max3185x.cpp:282
#6  0x0806ba42 in Max3185xRead::getMax3185xEgtValues (this=this@entry=0x200047a0 <instance>, ch=ch@entry=0, temp=temp@entry=0x20004a54 <instance+692>, coldJunctionTemp=coldJunctionTemp@entry=0x0)
    at ./hw_layer/sensors/max3185x.cpp:423
#7  0x08037ac4 in Max3185xRead::ThreadTask (this=0x200047a0 <instance>) at ./hw_layer/sensors/max3185x.cpp:113
#8  0x080320fa in ThreadController<400>::main (this=<optimized out>) at ./controllers/system/thread_controller.h:33
#9  0x08064f60 in chibios_rt::_thd_start (arg=<optimized out>) at ChibiOS/os/various/cpp_wrappers/ch.cpp:41
#10 0x08000b7a in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) frame 2
#2  0x08004c5a in spi_lld_start (spip=spip@entry=0x2000865c <SPID5>) at ChibiOS/os/hal/ports/STM32/LLD/SPIv1/hal_spi_lld.c:406
406	      osalDbgAssert(spip->dmatx != NULL, "unable to allocate stream")
```

dron0gus, 2024-05-12:
Actually both ADC1 and SPI5 need the same DMA channel:
```
#define STM32_ADC_ADC1_DMA_STREAM           STM32_DMA_STREAM_ID(2, 4)
#define STM32_SPI_SPI5_TX_DMA_STREAM        STM32_DMA_STREAM_ID(2, 4)
```

dron0gus, 2024-05-12:
https://github.com/rusefi/rusefi/pull/6506

pchmura4, 2024-05-13:
It's working fine now, we can close this out, thanks!

closed 2024-05-13 by pchmura4