rusefi/rusefi #6503: 154 hyuindai does not init 33810

rusefillc, 2024-05-12:
```
2024-05-12_11_58_13_899: EngineState: pin PA3 ((null)): AcceleratorPedalPrimary
2024-05-12_11_58_13_899: EngineState: pin PA4 ((null)): Tps1Primary
2024-05-12_11_58_13_899: EngineState: pin PA5 ((null)): BatteryVoltage
2024-05-12_11_58_13_899: EngineState: pin PA8 ((null)): ETB disable
2024-05-12_11_58_13_901: EngineState: pin PA11 ((null)): USB DM
2024-05-12_11_58_13_908: EngineState: pin PA12 ((null)): USB DP
2024-05-12_11_58_13_909: EngineState: pin PA15 ((null)): spi3-cs-etb
2024-05-12_11_58_13_910: EngineState: pin PB0 ((null)): Tps1Secondary
2024-05-12_11_58_13_912: EngineState: pin PB3 (X8 AuxLS1): SPI clock
2024-05-12_11_58_13_914: EngineState: pin PB4 ((null)): SPI master in 
2024-05-12_11_58_13_914: EngineState: pin PB5 ((null)): SPI master out
2024-05-12_11_58_13_915: EngineState: pin PB6 ((null)): SD CS
2024-05-12_11_58_13_916: EngineState: pin PB13 ((null)): SPI clock
2024-05-12_11_58_13_917: EngineState: pin PB14 ((null)): SPI master in 
2024-05-12_11_58_13_918: EngineState: pin PB15 ((null)): SPI master out
2024-05-12_11_58_13_922: EngineState: pin PC0 ((null)): MapSlow
2024-05-12_11_58_13_926: EngineState: pin PC2 ((null)): Clt
2024-05-12_11_58_13_928: EngineState: pin PC3 ((null)): Iat
2024-05-12_11_58_13_930: EngineState: pin PC4 ((null)): AcceleratorPedalSecondary
2024-05-12_11_58_13_932: EngineState: pin PC6 ((null)): ETB Enable
2024-05-12_11_58_13_934: EngineState: pin PC7 ((null)): ETB Dir 1
2024-05-12_11_58_13_935: EngineState: pin PC8 ((null)): ETB Enable
2024-05-12_11_58_13_937: EngineState: pin PC9 ((null)): ETB Dir 1
2024-05-12_11_58_13_941: EngineState: pin PC10 ((null)): SPI clock
2024-05-12_11_58_13_941: EngineState: pin PC11 ((null)): SPI master in 
2024-05-12_11_58_13_941: EngineState: pin PC12 ((null)): SPI master out
2024-05-12_11_58_13_941: EngineState: pin PC13 (Coil 1): mc33810 DIRECT IO
2024-05-12_11_58_13_941: EngineState: pin PD0 ((null)): CAN RX
2024-05-12_11_58_13_942: EngineState: pin PD1 ((null)): CAN TX
2024-05-12_11_58_13_942: EngineState: pin PD8 ((null)): spi3-cs-wg
2024-05-12_11_58_13_942: EngineState: pin PD10 (INJ_4 k28): mc33810 DIRECT IO
2024-05-12_11_58_13_942: EngineState: pin PD11 (INJ_3 k27): mc33810 DIRECT IO
2024-05-12_11_58_13_942: EngineState: pin PD12 (MIL): checkEnginePin
2024-05-12_11_58_13_942: EngineState: pin PD13 ((null)): mc33810 CS
2024-05-12_11_58_13_943: EngineState: pin PE3 (Coil 4): mc33810 DIRECT IO
2024-05-12_11_58_13_943: EngineState: pin PE4 (Coil 3): mc33810 DIRECT IO
2024-05-12_11_58_13_943: EngineState: pin PE5 (Coil 2): mc33810 DIRECT IO
2024-05-12_11_58_13_943: EngineState: pin PE7 ((null)): led: comm status
2024-05-12_11_58_13_943: EngineState: pin PE8 ((null)): led: warning status
2024-05-12_11_58_13_943: EngineState: pin PE12 (X9 Digital1): Trigger #1
2024-05-12_11_58_13_943: EngineState: pin PE13 (X7 Digital2): cam1
2024-05-12_11_58_13_943: EngineState: pin PE14 (X4 Digital3): cam2
2024-05-12_11_58_13_944: EngineState: pin PF8 (A29 Brake Input): brake pedal switch
2024-05-12_11_58_13_944: EngineState: pin PF9 (A57 AC Request): ac_switch
2024-05-12_11_58_13_944: EngineState: pin PF10 (A43 Clutch Input): clutch up switch
2024-05-12_11_58_13_944: EngineState: pin PF11 (K40 VSS): VehicleSpeed
2024-05-12_11_58_13_944: EngineState: pin PG0 ((null)): led: CRITICAL status
2024-05-12_11_58_13_944: EngineState: pin PG1 ((null)): led: running status
2024-05-12_11_58_13_944: EngineState: pin PG3 (Fan Relay Low): Fan Relay
2024-05-12_11_58_13_945: EngineState: pin PG6 ((null)): ETB disable
2024-05-12_11_58_13_945: EngineState: pin PG7 (INJ_1 k25): mc33810 DIRECT IO
2024-05-12_11_58_13_947: EngineState: pin PG8 (INJ_2 k26): mc33810 DIRECT IO
2024-05-12_11_58_13_951: EngineState: pin PG9 ((null)): mc33810 EN
2024-05-12_11_58_13_954: EngineState: pin PG13 (Fuel Pump K70): Fuel pump Relay
2024-05-12_11_58_13_959: EngineState: pin PG14 (Main Relay K64): Main Relay
2024-05-12_11_58_13_959: EngineState: ext mc33810.OUT1: Injector 1 diagnostic: open_load short_to_bat overtemp
2024-05-12_11_58_13_959: EngineState: ext mc33810.OUT2: Injector 2 diagnostic: open_load short_to_bat overtemp
2024-05-12_11_58_13_959: EngineState: ext mc33810.OUT3: Injector 3 diagnostic: open_load short_to_bat overtemp
2024-05-12_11_58_13_960: EngineState: ext mc33810.OUT4: Injector 4 diagnostic: open_load short_to_bat overtemp
2024-05-12_11_58_13_960: EngineState: ext mc33810.GD0: Coil 1 diagnostic: open_load short_to_gnd overload 
2024-05-12_11_58_13_960: EngineState: ext mc33810.GD1: Coil 2 diagnostic: open_load short_to_gnd overload 
2024-05-12_11_58_13_960: EngineState: ext mc33810.GD2: Coil 3 diagnostic: open_load short_to_gnd overload 
2024-05-12_11_58_13_960: EngineState: ext mc33810.GD3: Coil 4 diagnostic: open_load short_to_gnd overload 
2024-05-12_11_58_13_960: EngineState: Total pins used: 63
2024-05-12_11_58_13_967: EngineState: confirmation_pins:4
2024-05-12_11_58_13_967: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.531756416
2024-05-12_11_58_14_006: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.541806208
2024-05-12_11_58_14_113: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.555723200
2024-05-12_11_58_14_222: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.551403968
2024-05-12_11_58_14_332: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.542871488
2024-05-12_11_58_14_439: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.545050624
```

dron0gus, 2024-05-12:
@rusefillc please 'pins' with https://github.com/rusefi/rusefi/pull/6504

rusefillc, 2024-05-12:
@dron0gus 
```
2024-05-12_13_36_07_951: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.562122368
2024-05-12_13_36_08_064: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.560827264
2024-05-12_13_36_08_065: EngineState: pin PA3 ((null)): AcceleratorPedalPrimary Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_065: EngineState: pin PA4 ((null)): Tps1Primary Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_065: EngineState: pin PA5 ((null)): BatteryVoltage Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_065: EngineState: pin PA8 ((null)): ETB disable Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_067: EngineState: pin PA11 ((null)): USB DM Mode Alt AF10 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_077: EngineState: pin PA12 ((null)): USB DP Mode Alt AF10 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_079: EngineState: pin PA15 ((null)): spi3-cs-etb Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_081: EngineState: pin PB0 ((null)): Tps1Secondary Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_083: EngineState: pin PB3 (X8 AuxLS1): SPI clock Mode Alt AF5 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_083: EngineState: pin PB4 ((null)): SPI master in  Mode Alt AF5 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_13_36_08_086: EngineState: pin PB5 ((null)): SPI master out Mode Alt AF5 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_087: EngineState: pin PB6 ((null)): SD CS Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_089: EngineState: pin PB13 ((null)): SPI clock Mode Alt AF5 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_092: EngineState: pin PB14 ((null)): SPI master in  Mode Alt AF5 Push-Pull Pull-up Speed Very High IN 1 OUT 0
2024-05-12_13_36_08_094: EngineState: pin PB15 ((null)): SPI master out Mode Alt AF5 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_094: EngineState: pin PC0 ((null)): MapSlow Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_096: EngineState: pin PC2 ((null)): Clt Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_097: EngineState: pin PC3 ((null)): Iat Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_099: EngineState: pin PC4 ((null)): AcceleratorPedalSecondary Mode Analog AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_102: EngineState: pin PC6 ((null)): ETB Enable Mode Alt AF2 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_103: EngineState: pin PC7 ((null)): ETB Dir 1 Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_104: EngineState: pin PC8 ((null)): ETB Enable Mode Alt AF3 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_105: EngineState: pin PC9 ((null)): ETB Dir 1 Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_107: EngineState: pin PC10 ((null)): SPI clock Mode Alt AF6 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_110: EngineState: pin PC11 ((null)): SPI master in  Mode Alt AF6 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_13_36_08_111: EngineState: pin PC12 ((null)): SPI master out Mode Alt AF6 Push-Pull No pull Speed Very High IN 0 OUT 1
2024-05-12_13_36_08_112: EngineState: pin PD0 ((null)): CAN RX Mode Alt AF9 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_112: EngineState: pin PD1 ((null)): CAN TX Mode Alt AF9 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_112: EngineState: pin PD2 ((null)): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_113: EngineState: pin PD8 ((null)): spi3-cs-wg Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_113: EngineState: pin PD10 (INJ_4 k28): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_113: EngineState: pin PD11 (INJ_3 k27): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_113: EngineState: pin PD12 (MIL): checkEnginePin Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_113: EngineState: pin PD13 ((null)): mc33810 CS Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_114: EngineState: pin PE7 ((null)): led: comm status Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_13_36_08_114: EngineState: pin PE8 ((null)): led: warning status Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_13_36_08_114: EngineState: pin PE12 (X9 Digital1): Trigger #1 Mode Input AF0 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_13_36_08_114: EngineState: pin PE13 (X7 Digital2): cam1 Mode Input AF0 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_13_36_08_115: EngineState: pin PE14 (X4 Digital3): cam2 Mode Input AF0 Push-Pull Pull-up Speed Low IN 1 OUT 1
2024-05-12_13_36_08_115: EngineState: pin PF8 (A29 Brake Input): brake pedal switch Mode Input AF0 Push-Pull Pull-up Speed Low IN 0 OUT 1
2024-05-12_13_36_08_115: EngineState: pin PF9 (A57 AC Request): ac_switch Mode Input AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_13_36_08_116: EngineState: pin PF10 (A43 Clutch Input): clutch up switch Mode Input AF0 Push-Pull Pull-up Speed Low IN 0 OUT 1
[... cut ...]
2024-05-12_13_36_08_148: EngineState: Total pins used: 63
2024-05-12_13_36_08_148: EngineState: confirmation_pins:4
2024-05-12_13_36_08_171: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.559313792
2024-05-12_13_36_08_282: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.558365824
2024-05-12_13_36_08_394: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.563899968
2024-05-12_13_36_08_505: EngineState: mc33810 spi loopback test failed [65535] vBatt=11.557595264
```

dron0gus, 2024-05-12:
@rusefillc https://github.com/rusefi/rusefi/pull/6505

rusefillc, 2024-05-12:
> @rusefillc #6505

yes it did it!!!

weird inj diag but that's a totally different story

```
2024-05-12_18_36_15_294: EngineState: pin PE10 ((null)): EN Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_18_36_15_296: EngineState: pin PE12 (X9 Digital1): Trigger #1 Mode Input AF0 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_18_36_15_296: EngineState: pin PE13 (X7 Digital2): cam1 Mode Input AF0 Push-Pull Pull-up Speed Very High IN 1 OUT 1
2024-05-12_18_36_15_298: EngineState: pin PE14 (X4 Digital3): cam2 Mode Input AF0 Push-Pull Pull-up Speed Low IN 1 OUT 1
2024-05-12_18_36_15_302: EngineState: pin PF2 ((null)): ADC Mux Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_302: EngineState: pin PF8 (A29 Brake Input): brake pedal switch Mode Input AF0 Push-Pull Pull-up Speed Low IN 0 OUT 1
2024-05-12_18_36_15_302: EngineState: pin PF9 (A57 AC Request): ac_switch Mode Input AF0 Push-Pull No pull Speed Low IN 0 OUT 1
2024-05-12_18_36_15_302: EngineState: pin PF10 (A43 Clutch Input): clutch up switch Mode Input AF0 Push-Pull Pull-up Speed Low IN 0 OUT 1
2024-05-12_18_36_15_303: EngineState: pin PF11 (K40 VSS): VehicleSpeed Mode Input AF0 Push-Pull Pull-up Speed Low IN 1 OUT 1
2024-05-12_18_36_15_303: EngineState: pin PG0 ((null)): led: CRITICAL status Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_18_36_15_303: EngineState: pin PG1 ((null)): led: running status Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_18_36_15_308: EngineState: pin PG2 ((null)): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_308: EngineState: pin PG3 (Fan Relay Low): Fan Relay Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_310: EngineState: pin PG5 (Aux P68): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_312: EngineState: pin PG6 ((null)): ETB disable Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_312: EngineState: pin PG7 (INJ_1 k25): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_313: EngineState: pin PG8 (INJ_2 k26): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_316: EngineState: pin PG9 ((null)): mc33810 EN Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_318: EngineState: pin PG11 ((null)): mc33810 DIRECT IO Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_318: EngineState: pin PG13 (Fuel Pump K70): Fuel pump Relay Mode GP out AF0 Push-Pull No pull Speed Low IN 0 OUT 0
2024-05-12_18_36_15_320: EngineState: pin PG14 (Main Relay K64): Main Relay Mode GP out AF0 Push-Pull No pull Speed Low IN 1 OUT 1
2024-05-12_18_36_15_322: EngineState: ext mc33810.OUT1: Injector 1 diagnostic: open_load short_to_bat overtemp
2024-05-12_18_36_15_322: EngineState: ext mc33810.OUT2: Injector 2 diagnostic: Ok
2024-05-12_18_36_15_324: EngineState: ext mc33810.OUT3: Injector 3 diagnostic: open_load 
2024-05-12_18_36_15_325: EngineState: ext mc33810.OUT4: Injector 4 diagnostic: open_load 
2024-05-12_18_36_15_326: EngineState: ext mc33810.GD0: Coil 1 diagnostic: Ok
2024-05-12_18_36_15_328: EngineState: ext mc33810.GD1: Coil 2 diagnostic: Ok
2024-05-12_18_36_15_328: EngineState: ext mc33810.GD2: Coil 3 diagnostic: Ok
2024-05-12_18_36_15_330: EngineState: ext mc33810.GD3: Coil 4 diagnostic: Ok
```

closed 2024-05-12 by rusefillc