rusefi/rusefi #6508: A/C Temp Sensor Analog In (not really important but nice to have)

EA11R, 2024-05-13:
**Is your feature request related to a problem? Please describe.**

Currently the A/C system works generally well and reliable in rusefi connected to my F6A AmpBox 
however the issue doesn't lie with rusefi A/C control system
however with older Hitachi , Denso japanese Stock ECU systems in some japanese cars made between 1978~2000ish
some of them had a A/C amplifier box that the RUSEFI could hook up to and that box does the job completely however
those systems are very unreliable and will eventually fail due to their mounting location , what makes it worse is that some ecus
have the amplifier built in and A/C relay them like the EA21R suzuki ECU I'm currently working with and every K6A based kei car Nissan , Mitsubishi 
and AmpBoxes of many more sports cars like the AE86, z32,r32 and most 90's nissans and 90's Toyotas  

**Would we need new configuration fields?**

(FAIL POINT CALIBRATION )
Minimum Temp Cut = Anything above 0.6V would be realistic
Max Temp Cut =  anything up to VCC supply 
---------------------------
Since it's a temp sensor like any other on AUX1 , AUX2 channels 
calibration of TEMP sensor reading
---------------------------
A/C Cutout temp =  Range where A/C reaches near freezing lines
---------------------------
Re-Engage TEMP : (Lowest A/C temp + Hysteresis Value)
---------------------------
Action Delay : Halt time between last CUT and last Activation 
to prevent constant cutting and reactivation 
---------------------------
**Describe the solution you'd like**

1- Ability Complete delete of any sort of AMP box of older cars with simple Activation methods 
I see no reason to keep those old circuitry in place where they are mostly mounted on the evap
humidity give them a tendency to rot over time .
 
2- Adding Compressor Protection to the firmware.

3- Being able to wire IN A/C on a standalone without retaining Factory harness and sensors.

**Describe alternatives you've considered**

for my application :
Lua script I had made to read AUX temp volt
setAcDisabled(true) 
whenever temp < safe

other applications : 
[image]
https://www.youtube.com/watch?v=x1HcEUkp9q8
simply adding one of those standalone boards
or adding the extra micro controller will do the job fine

however I don't see any hardware or firmware restrictions that makes this not possible but I do see a sacrifice of a AUX input
that can be utilized as a gauge

ElDominio, 2024-05-16:
I'm sorry to tell you, but I think this ticket is pretty poor, while trying to communicate something useful.

There is something in big bold letters that throws everything off; a screenshot of some ebay board and a youtube video. The first statement "A clear and concise description of what the problem is."  was not followed and I don't know if the devs will enjoy sifting through it.

I do think we need A/C Temperature/Pressure checks to get decently working A/C, could you try simplifying this ticket to be more easy to digest? The fact that Nissan/Suzuki/Subaru/Desno do it some way and that some amplifier does something is not important. State what sensor needs to be compensated how and what kind of calibration it needs, anything else is superfluous

ElDominio, 2024-05-16:
I know how this works since I have written functions in lua for this exact purpose (which is why Lua function setAcDisabled exists)

I don't know why you closed this thread instead of cleaning it up a bit, but in any case, it can be done with Lua and I think it will be in base firmware in due time.

EA11R, 2024-05-16:
I have done it with LUA however I went with the cheap standalone cut board to keep the A/C running on stock functions without using 3 inputs for A/C

Inputs used in LUA script 

Case 1: pressure switch

Evap temp : AUX
Pressure switch :button 
A/C request: Button 

Case 2: pressure sensor

Evap temp : AUX

Pressure sensor :AUX 
(Requires sensor limit setting)
A/C request: Button 

I have it setup in a way where I use only one input to control the A/C as if right now however for deleting the thermistor box they might come in handy in a car without most factory internals I think 

How I made my way around was like this 

Added that board on my adapter 
Connected the pressure switch to an injector drive transistor ground 
Gave the output to the thermistor board and the input to activate that transistor is LS from ECU

Given this sequence I don't need to sacrifice other inputs for adding this extra functionality 

[image]

EA11R, 2024-05-17:
more examples of how denso use this system in their ecu's 
Toyota Alteeza 3S (BEAMS) ECU 
[image]
[image]

the Alteeza ECU has the A/C sensor , pressure switch , compressor out all built in the oem ecu as well without any external amplifier outside of the ECM enclosure