rusefi/rusefi #6516: Traction control not working

zed65, 2024-05-14:
Ignition timing drop do not work as intended.
See log at:
[https://rusefi.com/online/view.php?log=1391](url)

MLV and TS does not match (don't know if that's a problem but strange)
[image]
[image]

rusefillc, 2024-05-15:
@ggurov do you say you got it to work and the confusion is about which direction the value is applied?

ggurov, 2024-05-15:
no, i need to add two pwm values to drive the wheel sensors. i'll get around to it and test this

ggurov, 2024-05-16:
https://github.com/rusefi/rusefi/issues/6520 maybe related

closed 2024-08-05 by zed65

rusefillc, 2024-08-05:
@zed65 what's the status?