rusefi/rusefi #6523: Alternator voltage target table

ElDominio, 2024-05-16:
**Is your feature request related to a problem? 
I'm always frustrated when I want to lower voltage under different conditions but cannot because voltage target is one value

**Would we need new configuration fields?**
Chnge single voltage target value to a 5x5 table

**Describe the solution you'd like**
Chnge single voltage target value to a 5x5 table

**Describe alternatives you've considered**
Stay with a single target voltage which is. Abit sad

**Additional context**
Under high engine loads, it's good to lower target voltage to free up some torque, and in cruising conditions we can target slightly higher voltages for battery life

rusefillc, 2024-06-14:
@ElDominio what are the axises? I can guess but you better spell out

[image]

ElDominio, 2024-06-14:
Axes are:

X i s RPM
Y is Load

ElDominio, 2024-06-21:
X values are RPM, Y values are Load

ElDominio, 2024-07-12:
Remove the written target voltage, and only use table

[image]

PR #6719 (2024-07-16): get rid of unused `targetVBatt` setting
`Target(Volts)` setting on `Alternator Settings` area in Tuner Studio), closes #6523
commit e4d54b79 (2024-07-16): get rid of unused `targetVBatt` setting (`Target(Volts)` setting on `Alternator Settings` area in Tuner Studio), closes #6523
closed 2024-07-16 by kifir23917