rusefi/rusefi #6528: luaSoftSparkSkip/luaHardSparkSkip probably bug

rusefillc, 2024-05-17:
We may have a defect: In C++ binary "+" operator has higher priority higher then ternary "?:"
		 * operator. It means that below we check if (luaSoftSparkSkip + engineConfiguration->useHardSkipInTraction)
		 * expression (sum of float and bool values) is zero or not. It's pretty weird :(
		 
```
		engine->softSparkLimiter.setTargetSkipRatio(luaSoftSparkSkip + engineConfiguration->useHardSkipInTraction ? 0 : tractionControlSparkSkip);
		engine->hardSparkLimiter.setTargetSkipRatio(luaHardSparkSkip + engineConfiguration->useHardSkipInTraction ? tractionControlSparkSkip : 0);
```

rusefillc, 2024-05-17:
@kifir23917 I am surprised that compiler has not thrown an error? are there any relevant flags?

rusefillc, 2024-05-17:
@ggurov could this be relevant for current TC attempts?

kifir23917, 2024-05-17:
> @kifir23917 I am surprised that compiler has not thrown an error? are there any relevant flags?

IMHO it's impossible to forbid implicit conversion from float to bool because people loves writing code like:
```
if (!floatValue) { /*...*/ }
if (floatValue) { /*...*/ }
```

closed 2024-05-17 by rusefillc