rusefi/rusefi #6544: 4G93 sync problems

ElDominio, 2024-05-25:
**Describe the bug**
Some 4G93 (equivalent trigger to 4G63) has issues when cranking, but everything seems fine

**Hardware**
 - Board: custom hardware paralela
 - Engine info 4G93

**To Reproduce**
Steps to reproduce the behavior:
1. Set up car 
2. try to crank
3. sometimes will not sync

**Expected behavior**
Good sync behavior? VVT syncgapratio seems reasonable yet no sync is achieved _sometimes_; when we can decode VVT position it starts right up

**Additional context**
Engine running happily:
https://rusefi.com/online/view.php?log=1403

Engine not running:
https://rusefi.com/online/view.php?log=1404

rusefillc, 2024-05-25:
@ElDominio grrrr 100 data rate

note 0.824 while range is 0.2 to 0.8

please try custom VVG gaps like 2 to 5 and 0.2 to 0.9?

[image]

[image]

ElDominio, 2024-05-25:
we have succesfully started 10 times; override with custom gaps fixed the issue

closed 2024-05-25 by rusefillc

ElDominio, 2024-05-25:
So we tried on another car and it started up every time, easily started like 30 times.

We changed cam sensors on problematic car, same results, starts up easily.

My wonder is, how is this showing up on log? Why can we see cam sensor reading something reasonable yet have it be bad?

commit 8b44f01f (2024-05-25): 4G93 sync problems fix #6544