rusefi/rusefi #6554: Extremely erratic wideband reading (AEM X-Series over CAN)

ElDominio, 2024-05-27:
[testwithrevD.zip](https://github.com/rusefi/hellen154hyundai/files/15451464/testwithrevD.zip)

It shows wideband having basically the same issue as https://github.com/mck1117/wideband/issues/318 on my comment

What is this?

rusefillc, 2024-05-27:
@ElDominio if that's CAN bus AEM X-series I cannot think how this could have anything to do with rusEFI hardware. Either CANbus works or it does not

ElDominio, 2024-05-27:
could it be firmware? which is why i mentioned older firmware

ElDominio, 2024-05-27:
> Anything is possible
> 
> You are not helping by not telling if the rest of the CANbus like dash is working or not
> 
> You are not helping by mixing so many things. If you assume older vs newer firmware please compare same exact board with both older and newer FW.

Remember Forte dash is not CAN.

We tried old vs new firmware on revB, works okay.

The error is quite similar to the one I experienced on my board

In any case, 
log for revD on newest fw
https://rusefi.com/online/view.php?log=1407

log for revB on newest fw
https://rusefi.com/online/view.php?log=1408

I do not see any CAN errors on revD 
but I will have to get a genesis to test revD on to confirm