rusefi/rusefi #6557: Latest console updates to a vesion from 3 days ago while reporting correct date

ElDominio, 2024-05-27:
**Describe the bug**
Seems to me that the latest console build is reporting everything is from today when in reality the bubild is outdated

**Hardware**
 - Board: 4chan revB
 - Engine info 3A92

**To Reproduce**
Steps to reproduce the behavior:
1. Download latest console
2. update with firmware update button
3. see on TunerStudio that firmware is from 20240524 when today is 20240527

**Expected behavior**
TunerStudio reports 20240527 and firmware has latest updates

**Screenshots**
TunerStudio Reports:
[image]

ECU File says:
[image]

But included ini is from
[image]

**Additional context**
I tried forcing to use 0527 ini but tunerstudio refuses???? 
[image]

I think this is more tunerstudio issue

ElDominio, 2024-05-27:
[image]

alas it was not a tunerstudio issue

rusefillc, 2024-05-27:
Here in NJ I have the same issue

[image]

[image]

[image]

rusefillc, 2024-05-27:
but current compile date

[image]

rusefillc, 2024-05-27:
[image]

rusefillc, 2024-05-27:
I broke only that date around https://github.com/rusefi/rusefi/commit/748432a6598bc69bcc8bb5cc777d74010f4da3bc

commit 87663b06 (2024-05-27): Latest console updates to a vesion from 3 days ago while reporting correct date fix #6557
only:fix
closed 2024-05-27 by rusefillc