rusefi/rusefi #6566: Launch control should have a variable ignition cut BEFORE the main Hard cut

ElDominio, 2024-05-29:
**Is your feature request related to a problem? Please describe.**
Current cut method is very harsh since we just get to the cut RPM and disable ignition, then re-enable once we drop below the cutRPM

**Would we need new configuration fields?**
Initial Ignition Cut % - % of ignition events cut when we get into the launch window
Final Ignition Cut % Before launch - % of ignition events cut once we get past Smooth Retard End RPM

**Describe the solution you'd like**
I'd like to be able to start a NOT FULLY ignition cut before we hit the main RPM cut to make transition into launch cut less rough.
I think Smooth Retard End RPM will need to be renamed since it now affects ignition cut. Maybe "Launch Corrections End RPM"

**Describe alternatives you've considered**
Even smarter launch logic for another time

kifir23917, 2024-06-06:
@ElDominio, currently we have two ways to affect ignition skip ratio:
1. via Lua script (`setSparkSkipRatio` and `setSparkHardSkipRatio`)
2. via `Traction Ignition Skip` dialog:
[image]

To calculate target skip ratio current implementation sums skip ratio values from this two sources: `LUA_SKIP_RATIO` + `TRACTION_SKIP_RATIO`.

Now we are adding the third way to affect ignition skip ratio: via [`Initial Ignition Cut %`] and [`Final Ignition Cut % Before launch (%)`] settings on `Launch Control` dialog:
[image]

How should we calculate the target skip ratio from the three sources listed above? Should it be a simple sum: `LUA_SKIP_RATIO` + `TRACTION_SKIP_RATIO` + `PRELAUNCH_SKIP_RATIO`?

ElDominio, 2024-06-06:
Yes, and `PRELAUNCH_SKIP_RATIO` should never be non-zero unless we are EXPLICITLY in the launch state

kifir23917, 2024-06-06:
Does `we are EXPLICITLY in the launch state` mean that [`Enable Launch Control`] setting is `true`?

ElDominio, 2024-06-06:
1) It only affects/uses the HardSkipLimiter, nothing about SoftSkip
2) correct, if we are in launch mode (meaning enable Launch = true, and all prerequisite conditions are also true), skip ratio should be 1 when LaunchRPM < real RPM
3) it should be "Final Ignition Cut% Before Launch"

kifir23917, 2024-06-06:
> 3. it should be "Final Ignition Cut% Before Launch"

Shouldn't we add to "Final Ignition Cut% Before Launch" sum `LUA_SKIP_RATIO + TRACTION_SKIP_RATIO` as we do while rpm is between [`Launch Control Window(RPM)`] and  [Launch Corrections End RPM]?

ElDominio, 2024-06-06:
When I mentioned
"Final Ignition Cut% Before Launch"
I meant the one in your supplied screenshot, where it is a user-enterable value. 

The final math should be 
`LUA_SKIP_RATIO + TRACTION_SKIP_RATIO + PRELAUNCH_SKIP_RATIO`

ElDominio, 2024-06-13:
Only one issue is left;

when we finally go over Launch RPM(rpm), all hard cut methods should be disabled once we drop below Launch RPM(rpm).

here are the settings:
[image]

here is the log
[launchtest97perfectnt.zip](https://github.com/user-attachments/files/15828007/launchtest97perfectnt.zip)

[image]

We want ignition to reactivate at 2999rpm, with a 60% ignition cut at 0 degrees of timing, since we are over ((LaunchRPM-LaunchRPM Corrections End RPM) but under Launch RPM
a.k.a.
3000>*2999*>2940

We only want full hard cut OVER LaunchRPM

kifir23917, 2024-06-13:
> We want ignition to reactivate at 2999rpm, with a 60% ignition cut at 0 degrees of timing, since we are over ((LaunchRPM-LaunchRPM Corrections End RPM) but under Launch RPM
> a.k.a.
> 3000>2999>2940

> We only want full hard cut OVER LaunchRPM

Doesn't this contradict to #5610? There is the following written:

> If I set my launch RPM to 4500rpm, and my RPM Hysteresis to 200rpm and my ignition cut to ON, my engine should disable ignition at 4500rpm, and then come back to life at 4300RPM.

ElDominio, 2024-06-13:
> 
> Doesn't this contradict to #5610? There is the following written:
> 
> > If I set my launch RPM to 4500rpm, and my RPM Hysteresis to 200rpm and my ignition cut to ON, my engine should disable ignition at 4500rpm, and then come back to life at 4300RPM.

Yes but i'm pretty sure that was outdated, I had stated that this was a step, but the final behavior is as i jsut described

So yes, it contradicts that

kifir23917, 2024-06-13:
> Yes but i'm pretty sure that was outdated, I had stated that this was a step, but the final behavior is as i jsut described

I've [rolled back](https://github.com/rusefi/rusefi/pull/6616) hysteresis functionality.

commit 0ad03ead (2024-06-13): update CHANGELOG (closes #6566)
closed 2024-06-13 by rusefillc