rusefi/rusefi #6570: A/C Pressure sensor considerations and use

ElDominio, 2024-05-30:
**Is your feature request related to a problem? Please describe.**
Most cars have A/C pressure sensors that let us know if the A/C pressure is too high (to shut off the compressor to not get into an overpressure situation), or too low (meaning there is a gas leak and we should not enable the compressor). We should be able to interpret voltage values from this 0-5v sensor and use it to control compressor state.

**Would we need new configuration fields?**
 calibration values for A/C pressure transducer
A/C Pressure Low Point Voltage
A/C Pressure High Point Voltage
A/C Low Point value - in kpa
A/C High point value - in kpa
Pressure Low disable - value of A/C pressure in kpa that compressor is disengaged
Pressure high disable - ""

**Describe the solution you'd like**
Add these calibration points to disable the compressor if bounds are exceeded

ElDominio, 2024-06-14:
Also cousin of #4054

kifir23917, 2024-06-17:
@ElDominio, what are the minimum and maximum values in kPa that we want to use for `Pressure Low disable` and `Pressure high disable` settings?

ElDominio, 2024-06-17:
Bounds should be 0 and 500 kpa

Defaults should be 100 and 300 minimum maximum respectively

kifir23917, 2024-06-18:
The functionality is ready for manual testing:
[image]
[image]
Automated tests are still in work.

kifir23917, 2024-06-19:
@ElDominio, A/C controller uses **deadbands** to prevent rapid switching on/off of AC. To check if A/C pressure is out of limits I use **deadband** with width = 5 (as we so for `Max CLT(deg C)`, `Max TPS(%)`). It means that if `Pressure Low disable(kPa)` is `200 kPa`, than A/C will be turned off when pressure is less than `200 - 5 = 195 kPa` and turned on back only after pressure exceeds `200 + 5 = 205 kPa`. Is it OK or do we need to reduce deadband width for `Pressure Low disable(kPa)` and `Pressure High disable(kPa)` settings?

> Bounds should be 0 and 500 kpa

It's obvious that `Pressure Low disable(kPa)` setting will never turn off A/C at if its value does not exceed deadband width (`5` in current implementation). So I'm going to use `5 kPa` as minimal acceptable value for `Pressure Low disable(kPa)` setting. Does it make sense for you?

Zazins, 2024-06-19:
Hi!
Could you please add option with PWM Fan cotroll. As for now only can be controlled by PWM outputs and cant be set to on by AC. Its commonly used in cars to use PWM fans.
Thanks!

rusefillc, 2024-06-19:
@Zazins I would have to remove your comment simply because it does not belong on "A/C Pressure sensor consideration" ticket

@Zazins feel free to create a separate ticket

ElDominio, 2024-06-21:
ac pressure has no gauge, so no way to corroborate

kifir23917, 2024-06-23:
I've added gauges for A/C Pressure:
[image]

ElDominio, 2024-06-26:
Confirmed working with nice gauges! Can close

closed 2024-06-26 by rusefillc

ElDominio, 2024-08-13:
Hysteresis needs to be configruable! I have noticed a hardcoded hysteresis of 5kpa, i need to be able to decide if i want to wait X amount of KPA until reenable

rusefillc, 2024-08-13:
@ElDominio let me know if you expect me to reopen this ticket (I would rather not since it was technically a success) or if you need help creating new follow-up ticket?