rusefi/rusefi #6576: CUSTOM_INSTANT_MAP_DECODING

jankovalski, 2024-06-01:
On 20240101 which I used until today there was one occurence of 6899: CUSTOM_INSTANT_MAP_DECODING error on startup without any further side effects. Runtime Data Rate in TunerStudio was 230. When I upgraded to 20240418 I noticed drop in Runtime Data Rate to 170 and warning counter of 6899 is now 66 on startup. I removed `warning(ObdCode::CUSTOM_INSTANT_MAP_DECODING, "Invalid MAP at %f", instantVoltage);` from firmware/controllers/engine_cycle/map_averaging.cpp and the warning counter is now at 0 and Runtime Data Rate is back at 230. Of course there were never any issues with map reading.

// hopefully this warning is not too much CPU consumption for fast ADC callback

Well - it is.

rusefillc, 2024-06-01:
@dron0gus what do you think?

dron0gus, 2024-06-01:
Now last livedata entry is at 1744 byte offset (https://github.com/rusefi/rusefi/blob/4dfd2a371e3ecbe2abd3d86ad10a820147c5982c/firmware/tunerstudio/generated/rusefi_mre_f4.ini#L3410) So there is total 1744 bytes of livedata needs to be readed for each refresh.
While at 20240101 it was only 1444 bytes (https://github.com/rusefi/rusefi/blob/a53e62505fc4b3f99946083da2117a4418d3bbdd/firmware/tunerstudio/generated/rusefi_mre_f4.ini#L3160)
BLOCKING_FACTOR is still the same = 750 bytes.
So at 20240101 TS does two reads to get whole livedata, while now it needs three reads.
Each read is request-reply pair. Each read or write over USB happens in one frame. Each frame is 1 mS.
So reading 1444 bytes of livedata requires 4 frames = 4 mS = 250 Hz.
Reading 1744 bytes takes 6 frames = 6mS = 166 Hz.
There is PR pending https://github.com/rusefi/rusefi/pull/6535 allowing to have infinity BLOCKING_FACTOR/blockingFactor ie doing read of whole live data within single request. My tests show 500Hz refresh rate.

dron0gus, 2024-06-01:
Ok, may be I was wrong about read and reply happens on different USB frames. If MCU is fast enough it can handle write and prepare reply within one frame.
`mapAveragingAdcCallback` is called in IRQ context and block task execution. Rate of this callback is quite high and `warinig` function is quite heavy. So it creates significant CPU load causing it slower reaction to USB.

mck1117, 2024-06-16:
I'm not actually sure what that warning was even for....

But it isn't the cause of your change in TS data rate.

dron0gus, 2024-11-04:
https://github.com/rusefi/rusefi/pull/7037 should improve situation with refresh rate.

rusefillc, 2024-11-04:
[image]

closed 2024-11-04 by rusefillc