rusefi/rusefi #6579: Sensors with variable data (like analog 0-5v and CAN data) should have some sort of smoothing

ElDominio, 2024-06-03:
**Describe the bug**
Some senors, like in this case AFR,  have high variability. In this case, a V-Twin has twin wideband O2 sensors, but the widebands show the lean spots in between cylinders firing. This should be smoothed out by software as to not throw off anything that depends on these sensors for PID stuff like closed loop AFR (I am thinking this makes P-term quite unstable)

**Hardware**
 - Board: Harley PNP
 - Engine info Harely vtwin

**Expected behavior**
I have a filter to smooth out the data for Air-Fuel ratio shown on screenshot, making it read a more stable value

**Screenshots**
[image]

mck1117, 2024-06-03:
Those are not lean spots between firings. There's still exhaust in the pipe. The sensor also doesn't have anywhere near enough bandwidth to resolve individual cylinders at 6000 rpm.

Do you have any data to support that this variation isn't real?

mck1117, 2024-06-03:
I bet that variation is real, just look at your MAP!

ElDominio, 2024-06-03:
> Those are not lean spots between firings. There's still exhaust in the pipe. The sensor also doesn't have anywhere near enough bandwidth to resolve individual cylinders at 6000 rpm.
> 
> Do you have any data to support that this variation isn't real?

I'll grab the log from their stuff

This is a V Twin, with individual exhausts per cylinder (i did mention it being a vtwin), and MAP reads from one cylinder (i increased sample time to 150ms at 540 degrees). I think it's not totally unreasonable to think we can see one cylinder reflected on the sheet? But idk

mck1117, 2024-06-04:
> I think it's not totally unreasonable to think we can see one cylinder reflected on the sheet

But what other gas would there be in the pipe? How would there be something other than exhaust from the previous firing near the sensor?

mck1117, 2024-06-04:
> I think it's not totally unreasonable to think we can see one cylinder reflected on the sheet

6000 rpm = 100hz = 1 cylinder firing every 10ms

ElDominio, 2024-06-04:
assuming an pressure wave can travel at the speed of sound; 1,125ft/s (let's say slower for te sake of it, 1000ft/s), so 1ft/ms, on a 3 foot long pipe (this is a bike, where the exhaust is halfway of the chassis, probably about a foot long but 3ft to give it favorable numbers) the exhaust gas would leave the exhaust with about 15ms to spare (3foot pipe * 1ft/ms = 3ft of travel in 3ms), which I also think is ample  time for fresh air to come into the pipe and start finding its way into the sensing element of the 4.9

ElDominio, 2024-06-05:
Log had min/max values for AFR at 10/16, meaning small variations look enormous. Here is the log
[2024-06-03_14.41.44wot7.mlg.zip](https://github.com/user-attachments/files/15574507/2024-06-03_14.41.44wot7.mlg.zip)

I don't think it's that critical but idk

benassi124, 2024-06-05:
> Log had min/max values for AFR at 10/16, meaning small variations look enormous. Here is the log [2024-06-03_14.41.44wot7.mlg.zip](https://github.com/user-attachments/files/15574507/2024-06-03_14.41.44wot7.mlg.zip)
> 
> I don't think it's that critical but idk

3 out of the 7 units that have gone out so far, the customers have complained about this, they arent used to this, its been a hard sell on that aspect so far

ElDominio, 2024-06-05:
Being not used to it doesn't make it incorrect. What bad effects does the AFR not being a smooth line cause the bike? Powerband looks fine, response looks fine. 3 out of the 7 people working on them are used to old, slow unresponsive analog widebands that update 10 times per second at best

benassi124, 2024-06-05:
@ElDominio not sure if this helps at all, two exactly the same engines, the left chart is maxxecu, which is currently considered the gold standard, by hardcore harley/bike racers, on the right is our ECU.
The left is what most of these guys are expecting, I am nowhere near smart enough to know why the left with maxxecu is so much smoother and less jumpy, however, I am hoping to learn
[image]

ElDominio, 2024-06-05:
Can you provide the log itself from the MaxxECU?

benassi124, 2024-06-05:
[Suburban HD Bridgestone Bike_10539last_10545.zip](https://github.com/user-attachments/files/15596080/Suburban.HD.Bridgestone.Bike_10539last_10545.zip)

benassi124, 2024-06-05:
> [Suburban HD Bridgestone Bike_10539last_10545.zip](https://github.com/user-attachments/files/15596080/Suburban.HD.Bridgestone.Bike_10539last_10545.zip)

@ElDominio

benassi124, 2024-06-05:
1 CAN WBO2, 1 Analog. Lambda A on the left is analog, Lambda B on the right is CAN.

Im not as much worried about the lambda as I am to lambda reacting to those crazy fluctuations in MAP, I was trying to illustrate that its not an engine thing, but something else potentially.
Or maybe im dumb, which is also very likely but willing to learn lol

[image]

benassi124, 2024-06-05:
here is a log from a maxxecu without that odd-fire map filter
[image]

ElDominio, 2024-06-05:
I _think_ this is the purpose of MAP sampling angle and window; but we'd have to play around with it to fix the MAP reading so it looks less jumpy

benassi124, 2024-06-05:
> I _think_ this is the purpose of MAP sampling angle and window; but we'd have to play around with it to fix the MAP reading so it looks less jumpy

> I _think_ this is the purpose of MAP sampling angle and window; but we'd have to play around with it to fix the MAP reading so it looks less jumpy

Correct I agree, we could possibly code in different levels of map sampling so it can be user friendly or a "User Defined Setting" where you can manually set them for more advanced users such as yourself.

Because when uncle fester and his 3 hillbilly mechanics in the hills of west virgina get their hands on it theyre not going to sit there and dial it in lol. It will help overall UX of RusEFI

benassi124, 2024-06-10:
@ElDominio 
Adding basic smoothing factor of 5 made this log much more readable by a tuner imo, 

[image]

rusefillc, 2024-06-15:
@benassi124 sounds like you want the level of smoothness similar to https://rusefi.com/online/view.php?log=1405 ?

rusefillc, 2024-06-15:
@ElDominio @benassi124 shall we try different numbers up to 12?

[image]

rusefillc, 2024-06-15:
@benassi124 @ElDominio let's start with MAP exponential average

[image]

left has default value '1' while right has alpha of 0.4

[image]

zed65, 2024-06-19:
If you set the MAP exponential average Alpha to a bigger value than 1 it throws a CRC check error. Maybe a MAX value of 1 should be implemented?

rusefillc, 2024-06-19:
@zed65 yes https://github.com/rusefi/rusefi/commit/f7d5d48d00fe7b507ed88da66236193a295ffb89