rusefi/rusefi #6588: Scheduled timestamp something? Screenshot

ElDominio, 2024-06-07:
**Describe the bug**
Latest update, this error appears on key on

**Hardware**
 - Board: 4chan revI
 - Engine info 3A92

**To Reproduce**
Steps to reproduce the behavior:
1. Update to latest
2. see error on key on, only boots on properly on USB first

**Expected behavior**
Not have this error

**Screenshots**
[image]

rusefillc, 2024-06-07:
@ElDominio please link tune.

rusefillc, 2024-06-07:
```
#define TOO_FAR_INTO_FUTURE_US (10 * US_PER_SECOND)
#define TOO_FAR_INTO_FUTURE_NT US2NT(TOO_FAR_INTO_FUTURE_US)
```

@ElDominio is your primingDelay more than 10 seconds somehow?

rusefillc, 2024-06-07:
@ElDominio I've improved logging please update firmware and report how it looks now

ElDominio, 2026-02-23:
have efinitely not seen in a while