rusefi/rusefi #6595: data write failed between older project and newer firmware

rusefillc, 2024-06-09:
Looks like we are so used that we have stopped reporting  :(

[image]

rusefillc, 2024-06-09:
is this similar to https://github.com/rusefi/rusefi/issues/6116 or different story?

zed65, 2024-06-09:
I erased my board with STM32Cube and flashed with fresh build, when I connect with TS it wanted to update firmware.ini, did that and then it starts to load tune to board and I got the same error.
[image]

rusefillc, 2024-06-09:
@zed65 please attach your project folder - both main controller and current tune files

zed65, 2024-06-09:
This is probably not related but I also got this error even though I got a fuelpress sensor configured. After a reboot of the board the error is gone.
[image]

zed65, 2024-06-09:
Here is my folder (if that was the one you wanted).
[780rusEfi.zip](https://github.com/user-attachments/files/15751890/780rusEfi.zip)

rusefillc, 2024-06-09:
https://github.com/rusefi/rusefi/releases/tag/release_20240304

zed65, 2024-06-09:
If I do as you did (with hardware), everything is fine. Don't know what's going on :(

rusefillc, 2024-06-09:
> If I do as you did (with hardware), everything is fine. Don't know what's going on :(

i assume that it is some specific setting in the older project which triggers the problem

@zed65 @ggurov since we know that sometimes it does NOT happen - looks like the next step is to compare affected tune to new tune defaults and then keep reducing difference report until problem stops happening?

it could also be about the total amount of changes not any specific setting

zed65, 2024-06-09:
Strange, it only happens if I use the above specific project. I grabbed my newer project from my car and did the same test and I got no error.
[780rusEfiNew.zip](https://github.com/user-attachments/files/15753940/780rusEfiNew.zip)

rusefillc, 2024-06-10:
> This is probably not related but I also got this error even though I got a fuelpress sensor configured. After a reboot of the board the error is gone. 

@zed65 should be better now https://github.com/rusefi/rusefi/commit/f1e21697d665737ecf57eeda9c468ac95922376e

dron0gus, 2024-10-06:
Issue is still exist in master.

dron0gus, 2024-10-20:
May be a ChibiOS CDC ACM implementation bug https://forum.chibios.org/viewtopic.php?f=35&t=6395

dron0gus, 2024-11-03:
https://github.com/rusefi/ChibiOS/pull/48

rusefillc, 2024-11-03:
@dron0gus what's the current status with https://github.com/rusefi/rusefi/pull/7031 change? Do we have a proper fix? Do we have a partial fix which significantly reduces the frequency of the issue? Else?

dron0gus, 2024-11-04:
@rusefillc fixes issue in current scenario on my side. Was not able to reproduce any more.

commit 209ea941 (2024-11-04): data write failed between older project and newer firmware fix #6595
closed 2024-11-04 by rusefillc