rusefi/rusefi #6668: Immobiliser

wmrk, 2024-06-27:
We're planning to use RusEFI in a daily car project, so we need to more secured silutions. I'm ready to design a discrete immo-box for some transponder and add it support to EFI board software. They could communicate via CAN-bus. Will this feature be ok?

rusefillc, 2024-06-27:
A bit too vague at the moment, not sure what would this mean for rusEFI exactly?

sounds like a Lua script which stops engine unless it has received something via CANbus

wmrk, 2024-06-27:
Seems Lua is a good quick solution for an unsecured primitive protocol. But if we want more secured and customizable IMMO option, we need:
- encription functions for authority check
- ability to perform any actions from UI such as attach/detach immo module, transponder programming, etc.
- password security for these actions
and other

Moreover, Lua scripting will be overoaded by any other functions in my car such as a/c control, evap control, dashboard, abs communication and potentially others.

Anyways, my suggestion is to collect user's requests for this option while I'm designing and testing this device. If there will no requests, it will be secured until opening a source code =)

mck1117, 2024-06-27:
> We're planning to use RusEFI in a daily car project (disclaimer: not for public roads!), so we need for more secured silutions

But if not on public roads, why do you need an immo? This seems wholly out of scope. Nobody is going to get very far trying to steal a car with an obscure ECU, so even "security by obscurity" might be enough here.

There also isn't any technical reason you can't implement this in Lua.

mck1117, 2024-06-27:
> But if not on public roads, why do you need an immo?

Note, this is also why there's no cruise control. Not because it wouldn't be possible, but because the existence of it encourages on-road use.

wmrk, 2024-06-28:
@mck1117 sorry I'd prefer to avoid such philosophical issues and discussions, only technical =) Dropped this phrase from the description, it was a legal disclaimer only

ElDominio, 2024-07-13:
> But if not on public roads, why do you need an immo? This seems wholly out of scope. Nobody is going to get very far trying to steal a car with an obscure ECU, so even "security by obscurity" might [...]
I don't want my car to be stolen in sketchy puerto rican racetracks /discussion
> There also isn't any technical reason you can't implement this in Lua.
This i do agree with

Also about cruise control, i want to cruise at 25mph at the pit lane in the racetrack