rusefi/rusefi #6696: Starter Motor remains activated if the brake is released before the maximum activation time

ADVecu, 2024-07-10:
**Describe the bug**

Bug related to the management of the starter motor, if I put the condition that in order to activate the starter motor the brake must be activated, if for some reason the brake is removed the output of the starter motor remains activated permanently without paying attention to the maximum activation time.

**Hardware**

 - Board: Custom Hellen Board, Custom Firmware via fw-custom-example, 20240707
 - Engine info: Test Bench

**To Reproduce**

Configure the activation of the starter motor so that the brake pedal is required to be activated

1. Press the brake pedal
2. Activate the Start/stop button
3. Before the maximum activation time is reached, release the brake pedal.

**Expected behavior**

Deactivate the starter motor if the brake is not pressed

**Screenshots**

[image]

**Additional context**

The blue LED that is seen is the output that I configured to activate starter motor, and on the screen you can see the brake status

https://github.com/rusefi/rusefi/assets/65104023/3d38503a-5533-4d37-8b7d-246ea3d61be6

rusefillc, 2024-07-10:
@ADVecu do you mean the defect happens if brake is release while cranking?

ADVecu, 2024-07-10:
Correct, while cranking.

An update, after testing more thoroughly, as soon as the rpm signal is detected, the starter motor control output is deactivated as expected, the test I carried out was with 0 rpms, it is a rare case but it can occur, If for some reason no trigger events are detected, the starter motor may remain activated indefinitely.