rusefi/rusefi #6697: what MAP sensor noise is expected/acceptable on bench Alphax 4chan F4 rev G

rusefillc, 2024-07-10:
[f4-again.zip](https://github.com/user-attachments/files/16155156/f4-again.zip)

[image]

[image]

rusefillc, 2024-07-10:
uaefi much less noise?!

[uaefi.zip](https://github.com/user-attachments/files/16155184/uaefi.zip)

[image]

mck1117, 2024-07-10:
this test without 12v power is inaccurate

rusefillc, 2024-07-10:
with +12v 4chan logs look the same

[image]

[F4-external-12v.zip](https://github.com/user-attachments/files/16155431/F4-external-12v.zip)

rusefillc, 2024-07-10:
uaefi has no analog muxes while 4chan has it

weirdest scope trace so far is from op-amp output

[image]

[image]

and then it was gone, scope looks clean while TS shows noise

rusefillc, 2024-07-10:
See also https://github.com/rusefi/rusefi/issues/6376

benassi124, 2024-08-27:
Can we close this?

rusefillc, 2024-08-27:
> Can we close this?

Nope, this not resolved and I believe that it's important

benassi124, 2024-08-27:
It is important, I just didnt know if we had progress on this