rusefi/rusefi #6715: TunerStudio communication has been quite unstable lately

ElDominio, 2024-07-14:
**Describe the bug**
A lot fo units are having issues where they show up in device manager but TunerStudio connects/disconnects at random, and console is slow to connect/sometimes fails. Is there something that has changed about USB?

**Hardware**
 - Board: 4chan, h154

**To Reproduce**
Steps to reproduce the behavior:
1. Connect USB
2. See ECU in device manager
3. TS connects and disconnects
4. Console tries to connect but never does

**Expected behavior**
A successful reliable connection

dron0gus, 2024-07-14:
Can you please:
0. connect device causing problems
1. delete TunerStudioAppDebug.txt in TunerStudioProjects directory.
2. start TunerStudio
3. set Communication->Comm Debug Log
4. open project for this ECU, wait until something bad happens.
5. close TunerStudio.
6. upload TunerStudioAppDebug.txt to this ticket.

rusefillc, 2024-07-15:
Unfortunately I've noticed the same

Bike not running, TS just dies after some seconds of showing gauges.

[TunerStudioAppDebug.zip](https://github.com/user-attachments/files/16229435/TunerStudioAppDebug.zip)

dron0gus, 2024-07-15:
Not reproduced under Linux.

ElDominio, 2024-07-17:
[TunerStudioAppDebug.zip](https://github.com/user-attachments/files/16257881/TunerStudioAppDebug.zip)

rusefillc, 2024-07-24:
@dron0gus @ElDominio status update

my backend nice Windows 11 laptop "April" has very unreliable TS
at the same time same exact bench, same USB cable same ECU I have older Windows 7 laptop extremely reliable

@ElDominio how many devices do you have? Any chance you can correlate instability with laptop?