rusefi/rusefi #6733: first mc33810 SPI communication is not happy

rusefillc, 2024-07-21:


rusefillc, 2024-07-21:
```
2024-07-20_23_07_25_890: EngineState: ********************** UID=41001E:35315106:34313831 crc=142 ******************************
2024-07-20_23_07_25_890: EngineState: MCU rev=451 flashSize=1024
2024-07-20_23_07_25_890: EngineState: CH_CFG_ST_FREQUENCY=1000
2024-07-20_23_07_25_891: EngineState: ENABLE_PERF_TRACE=0
2024-07-20_23_07_25_892: EngineState: STM32_ADCCLK=27000000
2024-07-20_23_07_25_894: EngineState: STM32_TIMCLK1=108000000
2024-07-20_23_07_25_895: EngineState: STM32_TIMCLK2=216000000
2024-07-20_23_07_25_895: EngineState: STM32_PCLK1=54000000
2024-07-20_23_07_25_896: EngineState: STM32_PCLK2=108000000
2024-07-20_23_07_25_897: EngineState: PORT_IDLE_THREAD_STACK_SIZE=32
2024-07-20_23_07_25_897: EngineState: CH_DBG_ENABLE_ASSERTS=0
2024-07-20_23_07_25_898: EngineState: CH_DBG_SYSTEM_STATE_CHECK=1
2024-07-20_23_07_25_899: EngineState: CH_DBG_ENABLE_STACK_CHECK=1
2024-07-20_23_07_25_901: EngineState: EFI_LOGIC_ANALYZER=1
2024-07-20_23_07_25_901: EngineState: EFI_TUNER_STUDIO=1
2024-07-20_23_07_25_902: EngineState: EFI_SHAFT_POSITION_INPUT=1
2024-07-20_23_07_25_902: EngineState: EFI_INTERNAL_ADC=1
2024-07-20_23_07_25_903: EngineState: Last Reset Cause: Reset from NRST pin
2024-07-20_23_07_25_903: EngineState: readFromFlash 80C0000
2024-07-20_23_07_25_904: EngineState: Reading second configuration copy
2024-07-20_23_07_25_904: EngineState: readFromFlash 80E0000
2024-07-20_23_07_25_905: EngineState: pin_markUsed: EN on PE10
2024-07-20_23_07_25_905: EngineState: Turning board ON [start-up]
2024-07-20_23_07_25_905: EngineState: applyNonPersistentConfiguration()
2024-07-20_23_07_25_906: EngineState: initializeTriggerWaveform(TT_TOOTHED_WHEEL_60_2/8)
2024-07-20_23_07_25_906: EngineState: initializeTriggerWaveform(TT_HALF_MOON/18)
2024-07-20_23_07_25_907: EngineState: initializeTriggerWaveform(TT_HALF_MOON/18)
2024-07-20_23_07_25_908: EngineState: pin_markUsed: led: comm status on PE7
2024-07-20_23_07_25_908: EngineState: engine sniffer: No
2024-07-20_23_07_25_909: EngineState: engine sniffer size=300
2024-07-20_23_07_25_910: EngineState: pin_markUsed: spi3-cs-etb on PA15
2024-07-20_23_07_25_910: EngineState: pin_markUsed: spi3-cs-wg on PD8
2024-07-20_23_07_25_910: EngineState: pin_markUsed: mc33810 CS on PD13
2024-07-20_23_07_25_911: EngineState: *****************+ mc33810_add 280 +*******************
2024-07-20_23_07_25_911: EngineState: initAdcInputs()
2024-07-20_23_07_25_912: EngineState: pin_markUsed: ADC Mux on PF2
2024-07-20_23_07_25_913: EngineState: WARNING: C6899: Invalid MAP at 4.065299036
2024-07-20_23_07_25_914: EngineState: pin_markUsed: knock ch1 on PF7
2024-07-20_23_07_25_915: EngineState: pin_markUsed: SPI clock on PB3
2024-07-20_23_07_25_915: EngineState: pin_markUsed: SPI master out on PB5
2024-07-20_23_07_25_916: EngineState: pin_markUsed: SPI master in  on PB4
2024-07-20_23_07_25_917: EngineState: pin_markUsed: SPI clock on PC10
2024-07-20_23_07_25_917: EngineState: pin_markUsed: SPI master out on PC12
2024-07-20_23_07_25_918: EngineState: pin_markUsed: SPI master in  on PC11
2024-07-20_23_07_25_918: EngineState: pin_markUsed: mc33810 EN on PG9
2024-07-20_23_07_25_919: EngineState: pin_markUsed: mc33810 DIRECT IO on PG7
2024-07-20_23_07_25_919: EngineState: pin_markUsed: mc33810 DIRECT IO on PG8
2024-07-20_23_07_25_920: EngineState: pin_markUsed: mc33810 DIRECT IO on PD11
2024-07-20_23_07_25_920: EngineState: pin_markUsed: mc33810 DIRECT IO on PD10
2024-07-20_23_07_25_921: EngineState: pin_markUsed: mc33810 DIRECT IO on PG5
2024-07-20_23_07_25_921: EngineState: pin_markUsed: mc33810 DIRECT IO on PD2
2024-07-20_23_07_25_921: EngineState: pin_markUsed: mc33810 DIRECT IO on PG11
2024-07-20_23_07_25_922: EngineState: pin_markUsed: mc33810 DIRECT IO on PG2
2024-07-20_23_07_25_922: EngineState: pin_markUsed: SD CS on PB6
2024-07-20_23_07_25_923: EngineState: extiTriggerTurnOnInputPin Trigger #1 PE12
2024-07-20_23_07_25_924: EngineState: pin_markUsed: Trigger #1 on PE12
2024-07-20_23_07_25_924: EngineState: extiTriggerTurnOnInputPin cam1 Intake Cam - K62
2024-07-20_23_07_25_925: EngineState: pin_markUsed: cam1 on Intake Cam - K62
2024-07-20_23_07_25_925: EngineState: extiTriggerTurnOnInputPin cam2 Exhaust Cam - K15
2024-07-20_23_07_25_926: EngineState: pin_markUsed: cam2 on Exhaust Cam - K15
2024-07-20_23_07_25_927: EngineState: pin_markUsed: Injector 1 on INJ_1 k25
2024-07-20_23_07_25_928: EngineState: pin_markUsed: Injector 2 on INJ_2 k26
2024-07-20_23_07_25_928: EngineState: pin_markUsed: Injector 3 on INJ_3 k27
2024-07-20_23_07_25_928: EngineState: pin_markUsed: Injector 4 on INJ_4 k28
2024-07-20_23_07_25_929: EngineState: pin_markUsed: Coil 1 on Coil 1
2024-07-20_23_07_25_929: EngineState: pin_markUsed: Coil 2 on Coil 2
2024-07-20_23_07_25_930: EngineState: pin_markUsed: Coil 3 on Coil 3
2024-07-20_23_07_25_930: EngineState: pin_markUsed: Coil 4 on Coil 4
2024-07-20_23_07_25_930: EngineState: pin_markUsed: checkEnginePin on MIL
[... 22 lines cut ...]
2024-07-20_23_07_25_940: EngineState: pin_markUsed: ETB Enable on PC8
2024-07-20_23_07_25_940: EngineState: pin_markUsed: led: warning status on PE8
2024-07-20_23_07_25_940: EngineState: pin_markUsed: led: running status on PG1
2024-07-20_23_07_25_941: EngineState: Running main loop
2024-07-20_23_07_25_941: EngineState: WARNING: P2135: Sensor fault: Tps1 is inconsistent
2024-07-20_23_07_25_942: EngineState: WARNING: P2137: Sensor fault: AcceleratorPedal is inconsistent
[... cut ...]
2024-07-20_23_07_25_946: EngineState: WARNING: P0264: Injector 2 fault: open_load 
2024-07-20_23_07_25_947: EngineState: WARNING: P0270: Injector 4 fault: open_load 
2024-07-20_23_07_25_947: EngineState: WARNING: P0353: Ignition 3 fault: open_load 
2024-07-20_23_07_25_948: EngineState: WARNING: P0354: Ignition 4 fault: open_load overload 
2024-07-20_23_07_25_948: EngineState: WARNING: P0351: Ignition 1 fault: short_to_gnd 
2024-07-20_23_07_25_948: EngineState: Priming pulse mass: 0.1278 g
2024-07-20_23_07_25_949: EngineState: Firing priming pulse of 17.14 ms
```

rusefillc, 2024-07-21:
[154.zip](https://github.com/user-attachments/files/16325284/154.zip)

rusefillc, 2024-07-21:
```
2024-07-21_12_38_12_556: EngineState: mc33810 spi loopback test failed [first 0x800F][spi check 0x400F][COR] vBatt=11.053919792 count=1
2024-07-21_12_38_12_556: EngineState: mc33810 spi loopback test failed [first 0x000F][spi check 0x400F][COR] vBatt=11.053919792 count=2
2024-07-21_12_38_12_557: EngineState: mc33810 Successful Init
```

[154-2.zip](https://github.com/user-attachments/files/16325301/154-2.zip)

Qwerty-OFF, 2024-07-21:
[image]
[image]

rusefillc, 2024-07-21:
[154-with-hack.zip](https://github.com/user-attachments/files/16325684/154-with-hack.zip)

closed 2025-06-03 by rusefillc