rusefi/rusefi #6748: PUBLIC_API_WEAK or BOARD_WEAK

rusefillc, 2024-07-26:
PUBLIC_API_WEAK or BOARD_WEAK same thing? let's pick which one stays and have a global replace

kifir23917, 2024-07-27:
It is sefe enough to get rid of one of these macrodefinitions in `rusefi/rusefi` repo, becaus in other repos `PUBLIC_API_WEAK` and `BOARD_WEAK` usages [are commented](https://github.com/search?q=BOARD_WEAK+OR+PUBLIC_API_WEAK+user%3Arusefi+-repo%3Arusefi%2Frusefi+&type=code).

kifir23917, 2024-07-27:
The difference between `PUBLIC_API_WEAK` and `BOARD_WEAK` is that `BOARD_WEAK` is empty when not `EFI_SIM_IS_WINDOWS`. I [tried](https://github.com/rusefi/rusefi/pull/6750/files) to make `BOARD_WEAK` equal to `PUBLIC_API_WEAK` and [got](https://github.com/rusefi/rusefi/actions/runs/10124394154/job/27998692514) a problem with `rusefi_simulator.exe`:
```
Linking build/rusefi_simulator.exe
/usr/bin/i686-w64-mingw32-ld: build/obj/engine_configuration.o: in function `Z17loadConfigurationv':
/home/runner/work/rusefi/rusefi/simulator/../firmware/controllers/algo/engine_configuration.cpp:701: undefined reference to `setBoardConfigOverrides()'
/usr/bin/i686-w64-mingw32-ld: build/obj/engine_configuration.o: in function `Z31getBoardMetaLowSideOutputsCountv':
/home/runner/work/rusefi/rusefi/simulator/../firmware/controllers/algo/engine_configuration.cpp:1117: undefined reference to `getBoardMetaOutputsCount()'
/usr/bin/i686-w64-mingw32-ld: build/obj/tunerstudio.o: in function `ZN11TunerStudio23handleWriteChunkCommandEP13TsChannelBasettPv':
/home/runner/work/rusefi/rusefi/simulator/../firmware/console/binary/tunerstudio.cpp:231: undefined reference to `setBoardConfigOverrides()'
/usr/bin/i686-w64-mingw32-ld: build/obj/tunerstudio.o: in function `ZN11TunerStudio23handleWriteValueCommandEP13TsChannelBaseth':
/home/runner/work/rusefi/rusefi/simulator/../firmware/console/binary/tunerstudio.cpp:336: undefined reference to `setBoardConfigOverrides()'
/usr/bin/i686-w64-mingw32-ld: build/obj/can_bench_test.o:/home/runner/work/rusefi/rusefi/simulator/../firmware/controllers/can/can_bench_test.cpp:65: undefined reference to `getBoardMetaOutputsCount()'
collect2: error: ld returned 1 exit status
make[1]: *** [../firmware/ChibiOS/os/common/startup/SIMIA32/compilers/GCC/rules.mk:175: build/rusefi_simulator.exe] Error 1
make[1]: Leaving directory '/home/runner/work/rusefi/rusefi/simulator'
make: *** [bundle.mk:132: ../simulator/build/rusefi_simulator.exe] Error 2
Error: Process completed with exit code 2.
```

rusefillc, 2024-07-27:
Something is pretty weird :( https://github.com/rusefi/rusefi/pull/6752

closed 2024-07-27 by rusefillc