rusefi/rusefi #6763: Improve Closed Loop Boost by adding a delay timer before PID starts adjusting

ElDominio, 2024-08-02:
**Is your feature request related to a problem? Please describe.**
Currently, boost PID starts working as soon as boost conditions are met; adding a delay timer until PID is able to adjust would help in cases where the turbo takes some time to spool,  since PID wants to keep adding duty to reach the target even though it is impossible

**Would we need new configuration fields?**
Yes, a Closed Loop Boost PID Delay where you can enter from 0 to 5.0 ms

**Describe the solution you'd like**
Add a delay timer to be able to delay PID adjustments

**Additional context**
[image]

LinkECU has something like this
[image]

The picture has descriptions, I can explain anything not understood. I only include it as an example, I think adding the delay timer gets us 90% of the way there, while they have a more complex control for little payoff (in my opinion)

ElDominio, 2025-01-18:
@FDSoftware podrias?

ZayTuned, 2025-02-15:
I came here to request a new feature but this is related to what I was looking for. The closed loop pids come on before the turbo is able to spool and maxes the available adjustment duty cycle then overshoots. I would love to have a setting where you can enter a boost threshold in relation to the target. For example your boost target is 200kpa, you can set a threshold where within X kpa of target the pids will active. Say 10-20kpa below the target