rusefi/rusefi #6783: Nitrous control

ElDominio, 2024-08-08:
**Is your feature request related to a problem? Please describe.**
I would like to be able to control nitrous systems from the ECU itself, to be able to compensate fuel/timing and control activation from the ECU 

**Would we need new configuration fields?**

1. Arming Method- this should have two choices: Digital Switch Input (a switch input like Launch Control), or a Lua Gauge (with > or < where a certain value over or under ARMS the system). The purpose of arming the nitrous control system is that we can use the car WITHOUT nitrous when the system is NOT armed, WITH nitrous when it IS armed.
2. Nitrous Relay Output - control the relay which activates the nitrous solenoid
3. Minimum TPS - minimum TPS for Nitrous Activation, set to 0 to disable
4. Minimum CLT - minimum CLT for Nitrous Activation, set to 0 to disable
5. Maximum MAP - maximum MAP for nitrous activation (for turbo cars that want to shut off nitrous once they spool), set to 0 to disable
6. Maximum AFR - maximum AFR for nitrous activation, set to 0 to disable
7. Activation RPM - at the RPM that the nitrous system engages when all conditions are met
8. Deactivation RPM -at the RPM that the nitrous system engages when all conditions are met
9. Deactivation window - this value is SUBSTRACTED from Deactivation RPM to dissalow nitrous from re-engaging once the deactivation RPM is reached. Meaning if Deactivation RPM = 8000 and Deactivation window = 1000, the nitrous system CANNOT re-engage until we fall below 7000RPM (8000-1000).
10. Fuel adder % - this is a % of fuel which is added to the base main fuel, since we are basically injecting additional air
11. Ignition retard - this is timing REMOVED from actual final timing (after all corrections) due to additional air

**Describe the solution you'd like**
Add the stuff up there.

**Describe alternatives you've considered**
Using Lua but this allows others to use it

**Additional context**
This is how it looks in other projects
[image]

benassi124, 2024-08-09:
I also need this! Thank you

ElDominio, 2024-08-31:
behavior is similar to A/C control in the sense that we have limits to engage/disengage the output, and it is only on/off control

rusefillc, 2024-08-31:
See also https://github.com/rusefi/rusefi/issues/5704
See also https://github.com/rusefi/rusefi/issues/3449

benassi124, 2024-11-22:
@ElDominio Is the RPM deactivation window how we are accounting for a shift, to not spray through shift essentially? 

I also believe there should a toggle or timer so we are not spraying until launch control activation input is no longer active if using launch control

kifir23917, 2024-12-01:
@ElDominio, could you please specify desired default values for configuration fields listed in the issue description?

ElDominio, 2024-12-02:
Arming Method- NONE
Nitrous Relay Output - NONE
Minimum TPS - 80
Minimum CLT - 60
Maximum MAP - 0
Maximum AFR - 14
Activation RPM - 4000
Deactivation RPM - 7000
Deactivation window - 500

kifir23917, 2024-12-05:
> I also believe there should a toggle or timer so we are not spraying until launch control activation input is no longer active if using launch control

@ElDominio, is my understanding correct that we should prevent simultaneous working of Launch Control and Nitrous Control functionality? I mean that we shouldn't activate nitrous solenoid (as well as add fuel and correct ignition retard) if Launch Control is activated?
Should we deactivate nitrous solenoid on Launch Control activation?
Should Pre-Launch mode affect nitrous control?

benassi124, 2024-12-05:
Typically we would want to disable nitrous during launch control and then maybe do a "time after launch" for activation 

Also in dabbling with Nitrous Control, I noticed we are missing a field for Nitrous solenoid Relay output, typically people PWM a solid state relay so they can control Solenoid DC% and spray the full shot.

@ElDominio  I propose we also add the following fields as well

**PWM frequency**
The output frequency to be used in the Nitrous PWM stage.

**Opening duty**
The duty at which the solenoid starts to open. Used to compensate for solenoid dead time this could be the duty cycle % we revert to after launch control deactivation. 

**Solenoid Duty table**
Specifies the PWM duty % to be used to control the solenoid, should look similar to a blend table, **x-axis** to be rpm, **y-axis** to be user selectable, GEAR, TPS etc. and **z-axis** to be Solenoid Relay Output DC%

ElDominio, 2024-12-05:
We are jumping the gun here, the idea is to get a switched control working THEN implement frequency control/vairiations once SIMPLE control is achieved

kifir23917, 2024-12-06:
@ElDominio, I think that I've completed this functionality:

[image]

Could you please test it?

ElDominio, 2025-01-18:
I will try to test this week. I did forget an additional condition

Minimum VSS - Minimum speed to engage nitrous (we don't want nitrous when we don't have traction)

ElDominio, 2025-01-18:
@FDSoftware

rusefillc, 2025-02-27:
> Minimum VSS - Minimum speed to engage nitrous (we don't want nitrous when we don't have traction)

https://github.com/rusefi/rusefi/issues/7525

closed 2025-02-27 by rusefillc

rusefillc, 2025-03-01:
[image]

rusefillc, 2025-08-17:
@davidchristie-dotcom this ticket is closed since done, please create separate new ticket