rusefi/rusefi #6798: 3/0 trigger syncs faster with Single Tooth than 6G72 Cam

ElDominio, 2024-08-11:
**Describe the bug**
On two cars with the same crank patterns, and probably the same crank/cam sensors, the one with a Single Tooth cam (modified by us) syncs up noticeably faster

**Hardware**
 - Board: both on F407 paralela-based plug and play
 - Engine info 6G72/6A12

**To Reproduce**
Steps to reproduce the behavior:
1. Set crank trigger to 3/0
2. Set cam to appropriate one

**Expected behavior**
at least similar sync times (6g72 should be faster, since we have 4 sync points to start if we check crank at the time a change is detected on cam, but at least as fast as single tooth

**Screenshots**
6G72 Cam:
[image]

Single tooth cam:
[image]

**Additional context**
Log for 6G72:
https://rusefi.com/online/view.php?log=1416

Logs for single tooth:
https://rusefi.com/online/view.php?log=1417

rusefillc, 2024-08-26:
assuming you have no noise in the wires you never have loss of sync on single cam

on the attached log the complex cam sensor totally struggles to stay synched

until this cam sensor is much happier the comparison makes no sense. we have to address constant re-sync, do we have a logic analyzer file for this struggle here?

ElDominio, 2024-08-26:
[image]

rusefillc, 2024-08-27:
@ElDominio note how cam syncs only 6 times and often does not sync at all? if I am reading this right

any chance that logic analyzer is some sort of perfect signal while logs are different conditions?

[image]

rusefillc, 2024-08-27:
happy reference old from a BQS

[image]

ElDominio, 2024-08-29:
[crankinglogicandmlvlogs.zip](https://github.com/user-attachments/files/16805566/crankinglogicandmlvlogs.zip)

newewst logs on the car

ElDominio, 2024-10-13:
While running is fine, i feel like sync is still slow, and it is also inaccurate. Timing jumps +/- 20 degrees which makes it very difficult to start.

rusefillc, 2024-10-13:
@ElDominio you are overwhelming me with actionable data not.

ElDominio, 2024-10-13:
Forgot to add "will test this tomorrow and try to log"

We already have logic analyzer of this pattern, but i was thinking of getting a normal log

ElDominio, 2024-10-13:
Also, just found this on FTO Workshop manual
[image]

ElDominio, 2024-10-23:
More data

https://www.maxxecu.com/webhelp/trigger_system-mitsubitshi_6g72.html

[why is it running like shit.zip](https://github.com/user-attachments/files/17483494/why.is.it.running.like.shit.zip)

i wish i knew why :(

ElDominio, 2024-10-23:
[29scynerror.zip](https://github.com/user-attachments/files/17483548/29scynerror.zip)

at 29.6s ish,  see trigger error rise, but triggerPrimaryRise seems logical? who caused this error?

ElDominio, 2024-10-24:
[dyinglogiclogs.zip](https://github.com/user-attachments/files/17512243/dyinglogiclogs.zip)

I took a bunch of logs of the car having difficulty with cranking, and specifically, in "diedoutlog2.logicdata" at about 12.5s it has a sync loss event that leads to shutdown of the engine.

ElDominio, 2024-10-24:
[image]
here is the car losing sync and dying

ElDominio, 2024-10-28:
Conversation about sync errors on pure 3-0 on 6G72 cam have been moved to #7002 , this ticket will be used when 3-0 with single tooth is running again

ElDominio, 2024-12-02:
Improved physical circuit, almost never loses sync now;
BUT
In cranking, toothgaps are very large.  think we need crank/not-cranking tooth gaps for these low resolution triggers like 4g63 and 6g72.

New ticket?

rusefillc, 2025-06-29:
@ElDominio what's the state of this ticket? is it rightly "waiting-for-test" and assigned to you?

ElDominio, 2025-06-29:
The car that uses this trigger is only recently roaming the streets again, and I need to update the hardware on it. Will follow up in two days

closed 2025-07-20 by rusefillc