rusefi/rusefi #6804: A/C Pressure needs to have configurable hysteresis

ElDominio, 2024-08-13:
**Describe the bug**
Right now the hysteresis is hard coded, +/-5kpa (meaning it  needs to go over 5kpa of maximum to disable, and under 5kpa o maximum to re-enable).
This should be user configurable, where the maximum IS the cutoff point (meaning the moment we hit maximum AC compressor is disabled), and then the hysteresis provides the window to re-enable

Minimum does not need this since it exists to save the compressor in case of a gas leak.

New config field:
A/C Pressure Enable Hysteresis - A/C pressure needed to drop under Maximum to re-enable the compressor

kifir23917, 2024-08-19:
@ElDominio , just to check that we are on the same page...
> Minimum does not need this since it exists to save the compressor in case of a gas leak.

Is my understanding correct that you do not want to have hysteresis functionality for `Pressure Low disable (kPa)` setting at all?
I mean that in case when this setting is `100 kPa` than:
- when `A/C Pressure Sensor` returns `99 kPa` we disable A/C
- if then `A/C Pressure Sensor ` returns `100 kPa` we enable A/C back?
And we do not care about frequent A/C enabling/disabling if A/C pressure oscillates near `Pressure Low disable (kPa)` = `100 kPa`?

ElDominio, 2024-08-20:
> @ElDominio , just to check that we are on the same page...
> 
> > Minimum does not need this since it exists to save the compressor in case of a gas leak.
> 
> Is my understanding correct that you do not want to have hysteresis functionality for `Pressure Low disable (kPa)` setting at all? I mean that in case when this setting is `100 kPa` than:
> 
> * when `A/C Pressure Sensor` returns `99 kPa` we disable A/C
> * if then `A/C Pressure Sensor ` returns `100 kPa` we enable A/C back?
>   And we do not care about frequent A/C enabling/disabling if A/C pressure oscillates near `Pressure Low disable (kPa)` = `100 kPa`?
> 
> Or do you mean that we can leave existing behavior for `Pressure Low disable (kPa)` setting as it is implemented now?

I did mean we can leave existing behavior for Pressure Low disable as is implemented now, thank you for the catch

kifir23917, 2024-08-21:
@ElDominio, on all existing devices `A/C Pressure Enable Hysteresis` will be `0.0`. But in case of new device do we want to use some non-zero default value for this setting on new devices?

ElDominio, 2024-08-21:
10 would be fine

So if 200 is max and hysteresis is 10, re-activation pressure would be 190

closed 2024-08-23 by rusefillc

kifir23917, 2024-08-23:
@ElDominio, I've finished work on this ticket, could you please test it?

ElDominio, 2024-08-23:
Literally just tested it lol, works perfectly!

commit 29485ad9 (2024-08-23): Implement default value for `A/C Pressure Enable Hysteresis` setting (closes #6804)