rusefi/rusefi #6832: Spark hardware latency feature

ZayTuned, 2024-08-20:
I would really appreciate a spark latency setting added like in megasquirt firmware. I have an "obd0" Honda running the factory distributor and wirhout any latency compensation I'm needing to add 6-7 degrees of ignition advance near redline in order to account for the ignition retard. This setting eliminates the issue and would be very helpful

rusefillc, 2024-08-20:
@ElDominio do you understand the nature of the ask here?

ElDominio, 2024-08-20:
Yes

Car that use camshaft-based position sensors are prone to having timing retard on its own since the belts stretches with RPM. To compensate for this, you need to add a delay to the spark output, usually expressed in ms/1000rpm (every 1000rpm you add for example 80ms to compensate for belt stretch)

This _technically_ should be added to the main trigger angle, since the reason timing retards is because our base trigger angle is wrong when the belt is stretched

ZayTuned, 2024-08-20:
> Yes
> 
> Car that use camshaft-based position sensors are prone to having timing retard on its own since the belts stretches with RPM. To compensate for this, you need to add a delay to the spark output, usually expressed in us/1000rpm (every 1000rpm you add for example 80ms to compensate for belt stretch)
> 
> This _technically_ should be added to the main trigger angle, since the reason timing retards is because our base trigger angle is wrong when the belt is stretched
> 
> _**EDIT**_
> 
> usec, not msec. Microseconds is the correct timebase

Since we are accounting for a delay in the system, I think the microseconds need to be subtracted from the firing event not added. If a delay was added that would retard the timing further

rusefillc, 2025-03-01:
pretty much at the moment everyone uses  "timing adjusted for drift"

we want to explicitly separate "true timing" from drift delay with is a plain RPM-based adjustment

rusefillc, 2025-03-01:
@FDSoftware desired calibration units are microseconds

ZayTuned, 2025-03-03:
Thank you guys for doing the hard work of implementing it!

FDSoftware, 2025-03-05:
@rusefillc 
> we want to explicitly separate "true timing" from drift delay with is a plain constant time adjustment. 50us-ish (leaving zero as default but we can go one byte value 0 to 250us I guess)

If I understood correctly, this setting would not have to be shown within `correctedIgnitionAdvance` gauge values, it will be applied after these corrections as the per-cylinder trims do

@dialedinmotorsports 
> Norman has the right idea, it should be a delay within trigger setting window as it is a fixed number in a per-use basis. This is the easiest way to implement initially so it at least offers more accurate timing control. Many other units out there do not have this and it is bad practice. :)

endgoal would be a 2D curve or base value + multiplier at X rpm as @ElDominio  says?

rusefillc, 2025-03-05:
@FDSoftware sounds like only one scalar setting to begin with?

ElDominio, 2025-03-05:
> @rusefillc
>
> If I understood correctly, this setting would not have to be shown within `correctedIgnitionAdvance` gauge values, it will be applied after these corrections as the per-cylinder trims [...]

2d curve is better, base*multiplier is usually confusing.

2d curve vs rpm, 5 cells imo

rusefillc, 2025-03-05:
@ElDominio if coil delay is about wires and magic why would rpm matter at all?

Looks like sometimes it's just one value not even a curve

ElDominio, 2025-03-05:
[image]

Note belt
Note rpm

FDSoftware, 2025-03-06:
@ZayTuned  || @dialedinmotorsports  || @ElDominio 
Can you try this when the nightly are generated? As a first interaction it is a fixed numerical field as @rusefillc said

rusefillc, 2025-04-03:
Fun fact: no one cars to test/provide feedback

closed 2025-04-03 by rusefillc

ggurov, 2025-04-03:
https://youtu.be/GrwNXZe60ao shows this feature working