rusefi/rusefi #6840: open loop boost control does not allow Y axis MAP values greater than 100, even when Y axis is set to MAP

ggurov, 2024-08-21:
**Describe the bug**
Boost control open loop table does not allow Y bins to be set greater than 100, yet allows Y axis to be set to MAP. The purpose of this would be to create a "valley" to vary boost control % based on boost in a preset manner.

**Hardware**
 - Board: doesn't matter
 - Engine info doesn't matter

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Boost control open loop'
2. change 'Y axis' to "MAP"
3. change top load bin to 101
4. See error

**Expected behavior**
Ability to set  Y bins to useful MAP values

**Screenshots**
[image]

**Additional context**
It may be silly and not work in practice... but it should still let me shoot myself in the foot.

rusefillc, 2024-08-22:
#4802

rusefillc, 2024-08-22:
Follow up separate issue https://github.com/rusefi/rusefi/issues/6843

closed 2024-08-22 by rusefillc