rusefi/rusefi #6841: AFR Protection: Protection for leaning out under high load

ElDominio, 2024-08-22:
**Is your feature request related to a problem? Please describe.**
This feature is related to reacting to lean conditions under high load, so I guess it is a problem

**Would we need new configuration fields?**
TPS% For Protection: TPS value for which we check to see if we are over our target safety AFR
MAP for Protection: same as above, but for MAP
Max RPM Limit: Limit for RPM if we are in a cut event.
AFR Maximum: Maximum AFR value we can be at before we apply a cut
Time for protection: time in ms we need to be over the Maximum AFR to trigger a cut event.

Cut conditions are removed once (MAP and TPS) OR RPM are under their limit values

**Describe the solution you'd like**
A cut that looks at AFR and checks if we are not too lean, to save engine from lean conditions

ggurov, 2024-08-23:
[image]

is lambda protection not what you're looking for ?

ElDominio, 2024-08-23:
It 100% is.... why is it not in the PROTECTIONS MENU!!!!!!!!!!!!!!!!!!
[image]

[image]
WHAT

ggurov, 2024-08-23:
also, watch out, that table is lambda difference, regardless of what's used for main, afr or lambda. 

lambda protection table is lambda, but it works and it's great.

closed 2024-08-23 by rusefillc