rusefi/rusefi #6842: Wideband failure detection/safety

ElDominio, 2024-08-22:
**Is your feature request related to a problem? Please describe.**
Widebandd sensors do fail, and when they do they give bad data, which can be very problematic if using closed loop fueling.

**Would we need new configuration fields?**
Not really besides an enable/disable Wideband O2 sanity checks

**Describe the solution you'd like**
For analog widebands, we should have any voltage under "low voltage" or over "high voltage" be considered an error,. For CAN widebands, usually we have a "this wideband is not correct" message sent by the controller

In both cases, we should disable closed loop fueling and throw a warning that activates the CEL light.

rusefillc, 2024-08-22:
@ElDominio I am extremely excited that great minds think alike /s

See https://github.com/mck1117/wideband/issues/320 from five days ago.

rusefillc, 2024-08-22:
@mck1117 @dron0gus FYI.

ElDominio, 2026-02-23:
i think this exists?