rusefi/rusefi #6847: MEMORY LEAK DETECTED are these for real or fake news?

rusefillc, 2024-08-23:
For instance see drama at https://github.com/rusefi/alphax-4chan/issues/186

ElDominio, 2024-08-23:
relevant script: 
`
canRxAdd(0x200)
setTickRate(100)
t = Timer.new()
t : reset()

vssSensor = Sensor.new("VehicleSpeed")
-- any value would be considered valid for three seconds
vssSensor : setTimeout(3000)
vssReal = 0

cycleTime = Timer.new()
cycleTime : reset()
acCycleCount = 0

flipCounter = 0

function getBitRange(data, bitIndex, bitWidth)
	byteIndex = bitIndex >> 3
	shift = bitIndex - byteIndex * 8
	value = data[1 + byteIndex]
	if (shift + bitWidth > 8) then
		value = value + data[2 + byteIndex] * 256
	end
	mask = (1 << bitWidth) - 1
	return (value >> shift) & mask
end

function getTwoBytesLSB(data, offset, factor)
	return (data[offset + 2] * 256 + data[offset + 1]) * factor
end

function onTick()

	-- sensor reads
	local CLTread = 50
	if getSensor("CLT") then
		CLTread = math.floor(getSensor("CLT") + 0.5)
	else
		CLTread = 50
	end
	local RPMread = math.floor(getSensor("RPM") + 0.5)
	local RPMhi = RPMread >> 8
	local RPMlo = RPMread & 0xff

	local mapSensor = math.floor(getSensor("MAP") + 0.5) -- 4a0 byte4
	local accPedal = math.floor(getSensor("AcceleratorPedal") + 0.5) -- 210 byte2

	-- ________________________________________________________
	-- payloads
	local can4a0payload = { 0x00, 0x00, 0x00, 0x00, mapSensor, 0x00, 0x00, 0x00 }
	local can255payload = { 0x3F, 0xFF, 0x3F, 0xFF, 0x3F, 0xFF, 0xFF, 0xFF }
	local can210payload = { 0x00, 0x00, accPedal, 0x00, 0x40, 0x00, 0x00, 0xFF }
	local can343payload = { 0x80, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
	local can349payload = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
	local can3acpayload = { 0x01, 0xFF, 0x82, 0xFF, 0x00, 0x00, 0x00, 0x00 }

	canFanNoSpeed = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
	canFanSpeedLo = { 0x20, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00 }
	canFanSpeedHi = { 0x00, 0x00, 0x64, 0x00, 0x00, 0x00, 0x00, 0x00 }
	canCLTpayload = { CLTread, 0x4C, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }

	if flipCounter < 2 then
		can212payload = { 0x00, 0x00, 0x00, 0x00, 0xAA, 0x66, 0x00, 0x00 }
		can312payload = { 0x49, 0xB5, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
	else
		can212payload = { 0x00, 0x00, 0x00, 0x00, 0x2A, 0x66, 0x00, 0x00 }
		can312payload = { 0x09, 0xB5, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
	end

	can415payload = { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }

	-- rpm fun stuff
	if t : getElapsedSeconds() < 2 then
		canRPMpayload = { 0x00, 0x40, 0x3E, 0x02, 0x00, 0x1F, 0xFF, 0x00 }
	else
		if flipCounter < 3 then
			canRPMpayload = { 0x00, RPMhi, RPMlo, 0x40, 0x01, 0x00, 0x00, 0x00 }
		else
			canRPMpayload = { 0x00, RPMhi, RPMlo, 0x00, 0x00, 0x00, 0x00, 0x00 }
		end
	end

	-- _____________________________________________________________
	-- transfers
	txCan(1, 0x210, 0, can210payload) -- 210 is  accel pedal pos and ID
	txCan(1, 0x4A0, 0, can4a0payload) -- 4A0 is only MAP
	txCan(1, 0x308, 0, canRPMpayload)
	txCan(1, 0x608, 0, canCLTpayload)
	-- txCan(1, 0x212, 0, can212payload)
	txCan(1, 0x312, 0, can312payload)
	txCan(1, 0x415, 0, can415payload)
	txCan(1, 0x255, 0, can255payload)
	txCan(1, 0x101, 0, { 0x02 })
	txCan(1, 0x343, 0, can343payload)
	txCan(1, 0x349, 0, can349payload)
	txCan(1, 0x3AC, 0, can3acpayload)

	timeSinceAc = getTimeSinceAcToggleMs() / 1000

	acState = getOutput("acCompressorState")

	if (timeSinceAc < 0.5) then
		cycleTime : reset()
	end
	acCycle = cycleTime : getElapsedSeconds()

	if (acState == 1) and (acCycle > 20) then
		setAcDisabled(1)
		cycleTime : reset()
	end
	if (acState == 0) and (acCycle > 5) then
		setAcDisabled(nil)
		cycleTime : reset()
	end
	--print(cycleTime : getElapsedSeconds())

	if RPMread > 400 then
		txCan(1, 0x215, 0, { 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x07, 0x00 })
		if CLTread < 60 then
			txCan(1, 0x415, 0, canFanNoSpeed)
		elseif CLTread >= 60 and CLTread < 95 then
			txCan(1, 0x415, 0, canFanSpeedLo)
		elseif CLTread >= 95 then
			txCan(1, 0x415, 0, canFanSpeedHi)
		else
			txCan(1, 0x415, 0, canFanSpeedHi)
		end
	else
		txCan(1, 0x415, 0, canFanNoSpeed)
	end

end

function onCanRx(bus, id, dlc, data)
	if id == 0x200 then
	vssReal = getBitRange(data, 8, 12) *.065
		vssSensor : set(vssReal)
		-- valFromSensor = getSensor("VehicleSpeed")

	 --print(vssReal)
	end
end`

mck1117, 2024-08-23:
Lua can often tip over and leave itself in an incompletely cleaned up state if the script fails to load due to an out-of-memory error, as you saw in that other issue. The ECU force cleans it up during unload so it doesn't actually leak any memory. So long as it only does this when you run out of memory, I wouldn't worry about it.

rusefillc, 2024-08-23:
@ElDominio also plz use ``luamemory`` in console on F4 and F7 to compare output

ElDominio, 2024-10-01:
[image]

If i reove the neutralSwitch line, it is ok

[image]

ElDominio, 2024-11-26:
[image]

max limit is 10,100 and i have 706 characters and i am still triggering memory limit?

rusefillc, 2024-11-26:
Let's assume that there is no leak. Looks like at the moment 'leak' is part of "not enough memory"

Anyone who is trying to use any significant Lua on 407 without spending extra $1 at least on 427 is making the mistake of not spending extra $1 to get 427

ElDominio, 2024-11-28:
Should then wording be more clear? "Leak" implies shit code, "not enough memory" sends a clear message about cheapness

And the Lua is for E46 BMW testing, I was using built-in CAN for E46 but it is missing AC and CLT is incorrectly scaled.

rusefillc, 2024-11-30:
@ElDominio here's some progress https://github.com/rusefi/rusefi/commit/13b9b085a16472ec2c8a97560f204bc9ea09cd4b

closed 2024-11-30 by rusefillc