rusefi/rusefi #6863: ConfigurationImageFile add header

rusefillc, 2024-08-27:
We have ConfigurationImageFile.java with readFromFile and saveToFile methods

Let's add a 16 bytes header
* 4 byte file format version, let's start with 0x239 as initial version
* 4 bytes for header size, always 16
* 4 bytes size
* 4 bytes crc32
* payload

rusefillc, 2024-08-30:
SerialSandbox
```
I 240830 131424.806 [ECU AutoDetectPort9] SerialAutoChecker - Got signature=rusEFI master.2024.08.30.hd81.3580009858  from serial=COM52
```

rusefillc, 2024-08-30:
in addition to how we just save a binary can we please also automatically save same binary into a zip file as well?

rusefillc, 2024-08-30:
See also shared_io binary_config_image

kifir23917, 2024-09-04:
Now when we press `Connect` button in console UI, console creates in its folder `current_configuration.binary_image` file. This file is zip archive with two files:
1. `Image.bin` with image binary representation
2. `ConfigurationImageMetaVersion0_0.yaml` with image metadata
The screenshot below illustrates structure of `ConfigurationImageMetaVersion0_0.yaml` file:
[image]

closed 2024-09-04 by kifir23917