rusefi/rusefi #6873: Remove a space to the generated board name

mirusu400, 2024-08-30:
### Current issue

On both `firmware/controllers/algo/generated_lookup_engine_configuration.h` and `firmware/controllers/algo/generated_lookup_meta.h`, They concatenate string using only `QUOTE`

However, it might be adding a space between both `ENGINE_GENERATED_NAME`, `SHORT_BOARD_NAME`, `ENGINE_GENERATED_EXT`.

for Example:
https://godbolt.org/z/zWzsKr9W4

This seems normally works on `arm-none-eabi` compiler , but other compilers such as clang might got an unexpected results:
```
<source>:12:10: fatal error: 'rusefi_generated_ f407-discovery .h' file not found
   12 | #include META_GENERATED_H_FILENAME
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
<source>:11:35: note: expanded from macro 'META_GENERATED_H_FILENAME'
   11 | #define META_GENERATED_H_FILENAME QUOTE(META_GENERATED_NAME SHORT_BOARD_NAME META_GENERATED_EXT)
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<source>:4:18: note: expanded from macro 'QUOTE'
    4 | #define QUOTE(x) Q(x)
      |                  ^~~~
<source>:3:14: note: expanded from macro 'Q'
    3 | #define Q(x) #x
      |              ^~
<scratch space>:4:1: note: expanded from here
    4 | "rusefi_generated_ f407-discovery .h"
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
Compiler returned: 1
```

So, We need to removing a space between  `ENGINE_GENERATED_NAME`, `SHORT_BOARD_NAME`, `ENGINE_GENERATED_EXT`.

But it seems to be a more complicated issue than I thought.
Putting a macro string in `#include` seems more complicated than I thought, and I've tried to fix this, but I'm failing to include except for the current method.

Reference:
https://stackoverflow.com/questions/39038867/string-concatenation-for-include-path
https://stackoverflow.com/questions/4743822/generate-include-file-name-in-a-macro

Relased PR:
#6869

rusefillc, 2026-09-02:
@mirusu400 hello from 2026! Any chance that you can check if that's still an issue?

mirusu400, 2026-10-16:
Hello @rusefillc , Sorry for late reply, but somewhat reason your reply is not in my mail inbox notification;

When I report the Issue I trying to resolve the trouble but cause of complicated/nested macro it is hard to fix it, so I guess we need more high-level predefined preprocessor...

closed 2026-10-16 by rusefillc