rusefi/rusefi #6874: VVL control (switched cam)

ElDominio, 2024-08-31:
**Is your feature request related to a problem? Please describe.**
I would like to be able to control nitrous systems from the ECU itself, to be able to compensate fuel/timing and control activation from the ECU 

**Would we need new configuration fields?**

1. VVL Output - control the relay which activates the nitrous solenoid
2. Minimum TPS - minimum TPS for Nitrous Activation, set to 0 to disable
3. Minimum CLT - minimum CLT for Nitrous Activation, set to 0 to disable
4. Activation RPM - at the RPM that the VVL system engages when all conditions are met
5. Deactivation RPM -at the RPM that the VVL system engages when all conditions are met
6. Fuel adder % - this is a % of fuel which is added to the base main fuel, since we are basically injecting additional air
7. Ignition retard - this is timing REMOVED from actual final timing (after all corrections) due to additional air

**Describe the solution you'd like**
Add the stuff up there.

**Describe alternatives you've considered**
Using Lua but this allows others to use it

rusefillc, 2024-08-31:
Variable Valve Lift? Nitrous?

ElDominio, 2024-08-31:
nitrous references removed, my mistake

ElDominio, 2025-01-18:
@FDSoftware  very similar to nitrous control

ElDominio, 2025-08-17:
Not even very similar, it could literally be a clone of Nitrous Control with everything that says Nitrous changed to VVL or Switched Cam

My FTO has MIVEC + Nitrous and this would make me very happy

ElDominio, 2025-08-17:
@rusefillc @FDSoftware