rusefi/rusefi #6878: Ignition Timing drifting

firebat-hacs, 2024-09-01:
timing is set to 10 degrees fixed in tunerstudio
which means the two lines should overlap (PIP - Crank wheel/blue signal and SPOUT - output to ignition box/orange signal)
but it drifts like 27 degrees per 1000rpm
[timing issue.zip](https://github.com/user-attachments/files/16830296/timing.issue.zip)

rusefillc, 2024-09-01:
zero RPM: ignition timing gauge shows selected fixed timing value as expected

[image]

rusefillc, 2024-09-01:
during cranking we do not show fixed timing, is this broken or expected?

[image]

rusefillc, 2024-09-01:
@firebat-hacs can you help by grabbing some release maybe from six months ago and checking if ignition: timing gauge used to show fixed value below cranking RPM threshold and above cranking RPM threshold?

rusefillc, 2024-09-01:
@firebat-hacs thank you this looks better now! pretty much before pulling the scope please confirm what the gauges are showing - this was not timing drift, this was totally ECU commanding dynamic timing

[image]

closed 2024-09-01 by rusefillc