rusefi/rusefi #6895: Ford Coyote Gen2 Intake VVT operation (forward+backward motion)

ElDominio, 2024-09-06:
**Is your feature request related to a problem? Please describe.**
Intake VVT in Gen2 Coyote does not rest in a "fully retarded" position, but literally at the halfway point. We need logic to do forward/reverse motion, which in this motor is achieved with different duty cycles on intake VVT.

**Would we need new configuration fields?**
Fastest Reverse Duty: The duty cycle at which VVT moves backwards from base the fastest. Usually around 35%
Slowest Reverse Duty: The Duty Cycle: The duty cycle at which VVT moves backwards from base the slowest. Usually around 45%
Slowest Forward Duty: The duty cycle at which VVT moves forwards from base the slowest. Usually around 55%
Fastest Forward Duty: The duty cycle at which VVT moves forwards from base the fastest. Usually around 65+%
Dropdown to select Coyote Gen2 in the direction dropdown

**Describe the solution you'd like**
Implement this and allow interpreting +/- target angles in VVT target

**Describe alternatives you've considered**
I do not have many, right now I only move forwards by using a minimum of 50% duty to a max of 90.

https://forums.linkecu.com/topic/21734-advice-on-getting-ford-coyote-gen-3-started/#comment-125556

https://youtu.be/fD3AOurBkIE?t=100

rusefillc, 2024-09-09:
@ElDominio how is this not PID? are you asking for feedforward for PID for VVT? are you sure it's needed?

rusefillc, 2025-08-10:
https://github.com/rusefi/rusefi/issues/8363 is better

closed 2025-08-10 by rusefillc