rusefi/rusefi #6924: Is this really too much LUA for STM32F407VGT6?

ElDominio, 2024-09-17:
**Describe the bug**
2024-09-17_17_05_02_275: EngineState: LUA MEMORY LEAK DETECTED: 32 bytes used after teardown
2024-09-17_17_05_02_275: EngineState: LUA loading script length: 160...

The script  is 5 lines. Is this too much???

**Hardware**
 - Board: custom paralela board
 
**Expected behavior**
Build is succesful
**Screenshots**
[image]

mck1117, 2024-09-17:
nowhere does it say you’re out of memory

ElDominio, 2024-09-18:
how is it leaking memory tho, why is it screaming at me

rusefillc, 2024-09-18:
Dup of #6924?

rusefillc, 2024-11-20:
@ggurov says something is very broken
```
2024-11-19_19_53_51_497: EngineState: LUA: Tearing down instance...
2024-11-19_19_53_51_497: EngineState: LUA MEMORY LEAK DETECTED: 4121536 bytes used after teardown
```

rusefillc, 2024-12-16:
probably fixed by https://github.com/rusefi/rusefi/pull/7177

closed 2024-12-16 by rusefillc