rusefi/rusefi #6935: ETB-Style VVT

ElDominio, 2024-09-20:
**Is your feature request related to a problem? Please describe.**
Soon I will be tackling a Veloster N. This engine has a DC motor for intake cam VVT. is there support for this? I think @mck1117 runs a Skyactiv X that operates like this

**Would we need new configuration fields?**
VVT Drive Type: chose between Solenoid or DC Motor

**Describe the solution you'd like**
Be able to drive an hbridge to control VVT.

mck1117, 2024-09-20:
it’s not a DC motor, it’s a BLDC, and it isn’t just like an ETB, it spins at engine speed and speeds up/slows down to control phase

rusefillc, 2024-09-20:
This ticket was repurposed for BLDC VVT because that's the best way to improve the ticket where @ElDominio has mixed his honest desires with else right in the description

See DC VVT https://github.com/rusefi/rusefi/issues/6937

ElDominio, 2024-09-20:
It could be just like that but i do not know, i do see a vvt that says EVVT and has two positives and two negatives for what I assume is higher current

rusefillc, 2024-09-20:
https://github.com/rusefi/rusefi_documentation/blob/master/OEM-Docs/Mazda/2020-mx5-ECU.pdf 

[image]

rusefillc, 2024-09-20:
https://forum.miata.net/vb/showthread.php?t=738495 says https://www.mx5manual.com/page.html?p=workshop&s=SC3326&docid=SM352698 

[image]

mck1117, 2024-09-20:
https://www.hexorcism.com/16ND/sh13006/

mck1117, 2024-09-21:
also, this same BLDC vvt is used in all skyactiv-G engines. Pretty much all Mazda post-2015 or so.

rusefillc, 2024-09-21:
@mck1117 any info on what are the six wires? Looks like on is ground, what about the other five?

rusefillc, 2024-09-21:
PS: do you still have your ND?

mck1117, 2024-09-21:
This is the only other relevant page I can find in the service manual. Dunno what the 4 wires do.

[image]

mck1117, 2024-09-21:
> PS: do you still have your ND?

https://www.facebook.com/marketplace/item/1009515067639109