rusefi/rusefi #6944: Norman says Traction Control ETB drop x and Y are still flipped?

rusefillc, 2024-09-22:
[image]

rusefillc, 2025-05-05:
skip ratio is an integer output, which basically makes it cut off all cylinders or none
no interpolation there 
the ignition skip table is also integer type, 0 decimal spaces, with a max of 1 that ignition skip expects 
the flipped x and y is a start, once that's fixed and anybody even attempts to put it on the bench to get anything useful out of it, it should be extra apparent other things that need fixing

rusefillc, 2025-05-05:
@FDSoftware first thing first we need a unit test :(

rusefillc, 2025-06-03:
@FDSoftware what about the other two tables?

```
	tcTimingDropTable.initTable(engineConfiguration->tractionControlTimingDrop, engineConfiguration->tractionControlSlipBins, engineConfiguration->tractionControlSpeedBins);
	tcSparkSkipTable.initTable(engineConfiguration->tractionControlIgnitionSkip, engineConfiguration->tractionControlSlipBins, engineConfiguration->tractionControlSpeedBins);
```

FDSoftware, 2025-06-04:
> [@FDSoftware](https://github.com/FDSoftware) are we in a happy place to fix the issue now?

yup, maybe we need new ticket for this?:

> once that's fixed and anybody even attempts to put it on the bench to get anything useful out of it, it should be extra apparent other things that need fixing

rusefillc, 2025-06-04:
@FDSoftware let's fix here first and then poke gurov for follow-up separately

rusefillc, 2025-06-06:
@FDSoftware you touch implementation reliability and common sense of which axis is which but this ticket originally is a defect/bug report stating that something is performing completely not as expected from user perspective? I would assume that @ElDominio does not care about table3D API?

rusefillc, 2025-06-15:
@ElDominio  @ggurov poke?

PR #8023 (2025-06-05): Traction control ETB & spark tests
* Traction control ETB is not fliped, updated the comment & test, both on firmware and TS we have correct X/Y axis (as defined on #6944 screenshot X => vehicle speed; Y => wheel slip)
* added test for tables on `Traction Control Timing drop` and `Traction Control Skip Ignition`
* newer test calculates the end of the table on compile time instated of fixed array length, to prevent more issues like: https://github.com/rusefi/rusefi/pull/7775

relevant TS template section:
```ini
table = tractionEtbTableTbl,  tractionEtb,  "Traction Control ETB drop",	1
    xBin [... cut]