rusefi/rusefi #6946: MRE fals to CH_CFG_SYSTEM_HALT_HOOK due to enabled MC33816

dron0gus, 2024-09-23:
Mre has `DDEFS += -DEFI_MC33816=TRUE` (why?)
But no `engineConfiguration->mc33816_cs`, `engineConfiguration->mc33816_rstb` and etc configured.
So `int Pt2001::init()` returns before `spiStart(driver, &spiCfg);` But driver task is started.
And when Vbatt rises above threshold `pt.initIfNeeded();` -> `Pt2001Base::restart()` is called, which tries to deselect() with null driver.

dron0gus, 2024-09-23:
```
Thread 1 "noname" received signal SIGTRAP, Trace/breakpoint trap.
chSysHalt (reason=reason@entry=0x806e328 <__func__.12.lto_priv.1> "spiUnselect") at ChibiOS/os/rt/src/chsys.c:201
201	  CH_CFG_SYSTEM_HALT_HOOK(reason);
(gdb) bt
#0  chSysHalt (reason=reason@entry=0x806e328 <__func__.12.lto_priv.1> "spiUnselect") at ChibiOS/os/rt/src/chsys.c:201
#1  0x080075d0 in spiUnselect (spip=<optimized out>) at ChibiOS/os/hal/src/hal_spi.c:154
#2  0x0805f674 in Pt2001::deselect (this=<optimized out>) at ./hw_layer/mc33816.cpp:74
#3  Pt2001Base::restart (this=<optimized out>) at ./libfirmware/pt2001/src/pt2001.cpp:447
#4  Pt2001::initIfNeeded() [clone .constprop.0] (this=<optimized out>) at ./hw_layer/mc33816.cpp:254
#5  0x0803a4dc in mc33_driver_thread (p=<optimized out>) at ./hw_layer/mc33816.cpp:278
#6  0x08000b7a in _port_thread_start () at ChibiOS/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S:201
```

closed 2024-09-23 by rusefillc

rusefillc, 2024-09-23:
@dron0gus MRE has SPI and test points, it's possible to hook old external MC33816 board to MRE

Well I guess now it's a test configuration?

dron0gus, 2024-09-23:
Any other board with EFI_MC33816=TRUE is affected by this issue. User can set such config that will lead to ecu hang on start.

rusefillc, 2024-09-23:
@dron0gus fixed by https://github.com/rusefi/rusefi/commit/e9db6ba07c291eb14058217caed249fef3de7a40?

commit e9db6ba0 (2024-09-23): MRE fals to CH_CFG_SYSTEM_HALT_HOOK due to enabled MC33816 fix #6946