rusefi/rusefi #6948: Board compilation broken on Windows

rusefillc, 2024-09-23:
**Describe the bug**
Mre config does not comoile under cygwin

**Hardware**
 - microRusEFI

**workaround**
WSL or real linux, or github actions

rusefillc, 2024-12-18:
@FDSoftware there is nothing we can do right?

```
// https://gcc.gnu.org/bugzilla/show_bug.cgi?id=90826 Weak symbol does not work reliably on windows
// https://sourceware.org/bugzilla/show_bug.cgi?id=9687 Weak symbols not working on mingw32
#if !defined(IS_WINDOWS_COMPILER) || !IS_WINDOWS_COMPILER
#define PUBLIC_API_WEAK_SOMETHING_WEIRD __attribute__((weak))
```

FDSoftware, 2024-12-18:
Considering that the bug has been unsolved for no more and no less than 14 years, I think the correct approach would be:
* drop cygwin support, explain to contributors that the current/supported alternatives are WSL with Ubuntu (since setup_linux_environment.sh works both on Ubuntu and WSL running Ubuntu), and for people who don't even want to install anything, fork & github actions as explained in: [c8b585d](https://github.com/rusefi/rusefi_documentation/commit/c8b585d33d4df217c6d255c03cedb930345fe665)
* revert the `PUBLIC_API_WEAK_SOMETHING_WEIRD` changes
* review and revert if necessary all occurrences of `ifeq ($(OS),Windows_NT)` in makefiles
* review address sanitizer in makefiles, it looks disabled for windows