rusefi/rusefi #6951: Combine PIN `signal` and PIN `mode` combo boxes in the single combo box with 6 options

kifir23917, 2024-09-27:
We often use combination of 2 comboboxes for PIN configuration. On different dialogs they are named incionsistently:

[image]

I could make sense to combine them in the single combobox with the following options:
- DEFAULT (normal)
- DEFAULT (inverted)
- PULLUP (normal)
- PULLUP (inverted)
- PULLDOWN (normal)
- PULLDOWN (inverted)

rusefillc, 2024-10-10:
@ElDominio

dron0gus, 2024-10-10:
May be checkbox for "Inverted"?

rusefillc, 2025-02-03:
@dron0gus can we start with just a couple of pins just to try this?

dron0gus, 2025-02-11:
Draft https://github.com/rusefi/rusefi/pull/7419 . Unit tests modifications need careful validation.