rusefi/rusefi #6972: hybrid injection: separate fuel pressure handling

rusefillc, 2024-10-05:
At the moment we have staged injection - two injectors per cylinder with adjustable ratio between injectors

in case of hybrid injection (GDI and MPI) we need different fuel pressures on different injectors

rusefillc, 2024-11-12:
getFuelDifferentialPressure

```
	FuelPressureLow, // in kPa
	FuelPressureHigh, // in kPa
	FuelPressureInjector,
```

rusefillc, 2024-11-12:
https://www.youtube.com/watch?app=desktop&v=sz8cqygvPC4

[image]

[image]

https://www.hella.com/techworld/us/Technical/Sensors-and-actuators/Checking-a-high-pressure-fuel-pump-69306/

[image]

rusefillc, 2024-11-12:
https://www.youtube.com/watch?v=MIDItBqX3SQ

kifir23917, 2024-11-16:
For primary injectors we allow user to configure the following settings:
- `Injector flow`
- `Injector flow compenstaion mode`
- `Injector reference pressure(kPa)`

[image]

For secondary injectors we allow user to configure only `Secondary injector flow`:

[image]

As I understand to support adequately hybrid injection (GDI and MPI) we need to implement
- `Secondary injector flow compenstaion mode`
- `Secondary injector reference pressure(kPa)`
settings in addition to the existing `Secondary injector flow` setting and to use these settings in mathematics for secondary injectors.

@ElDominio, are you agree?

ElDominio, 2024-11-16:
This is correct. They have very different pressures so this is reasonable

kifir23917, 2024-11-16:
Surprise-surprise :) Neither [Injector flow](https://github.com/search?q=repo%3Arusefi%2Frusefi%20injector_flow&type=code) nor [Secondary injector flow](https://github.com/search?q=repo%3Arusefi%2Frusefi+injectorSecondary_flow&type=code) parameters from the screenshots above are not used in our C++ code.

kifir23917, 2024-11-22:
I've implemented `Secondary injector flow compensation mode` and `Secondary injector reference pressure (kPa)` settings:

[image]

commit a36a8614 (2024-11-22): implement `FuelDifferentialPressureTest` (closes #6972)
closed 2024-11-22 by kifir23917