rusefi/rusefi #6973: hybrid injection: second dead time

rusefillc, 2024-10-05:
At the moment we have staged injection - two injectors per cylinder with adjustable ratio between injectors

in case of hybrid injection (GDI and MPI) we need separate dead time/injection lag curves for different injectors

rusefillc, 2024-11-12:
@ElDominio do we even need dead time for GDI?

ElDominio, 2024-11-12:
This was discussed, and I'm not sure a clear conclusion was reached. From what I understand, since we have amplifiers for driving the injectors, voltage should be constant, resulting in no need for voltage-related dead times. Is this true?

rusefillc, 2024-11-12:
voltage-related correction would be https://github.com/rusefi/rusefi/issues/7068

the question here is - do we have any injector lag for GDI injectors?

rusefillc, 2024-11-12:
```
	//setting "flat" 0.2 ms injector's lag time
	setArrayValues(engineConfiguration->injector.battLagCorr, 0.2);
```

rusefillc, 2024-12-18:
@kifir23917 close or else?