rusefi/rusefi #6978: Lua Digital gauge

ElDominio, 2024-10-09:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when I have a true/false input in Lua and I can't pass it to firmware (like CAN start request)

**Would we need new configuration fields?**
On functions that use digital inputs, there would be LuaSwitch, like there is LuaGauge for analog inputs.

**Describe the solution you'd like**
Add a LuaSwitch that is passed to main firmware from Lua on digital inputs (starter, clutch up/down)

**Describe alternatives you've considered**
Full control in Lua, but misses main firmware integration 

**Additional context**
AcRequest is passable from Lua, but a generic LuaSwitch would have handled this and any other case

rusefillc, 2024-10-10:
https://github.com/rusefi/rusefi/issues/3676

rusefillc, 2024-10-10:
https://github.com/rusefi/rusefi/issues/6658

rusefillc, 2024-10-10:
See also https://github.com/rusefi/rusefi/issues/6951

ElDominio, 2024-10-10:
A list of additional sets that would be useful

setStartRequest - to enable start request from CAN to fire a physical pin
setClutchDownState - set clutch as pressed
setLaunchRequest - sets launch control as armed

ElDominio, 2024-10-10:
setClutchUpState

Could be changed to "setClutchState(var)" where var could be
0- clutch down
1- clutch up 
Nil - clutch unknown