rusefi/rusefi #6983: Lua 'startCrankingEngine' method

rusefillc, 2024-10-13:
We need ``startCrankingEngine`` Lua function behaving similar to physical start button

rusefillc, 2024-10-13:
@ElDominio please test

ElDominio, 2024-10-15:
About startCrankingEngine, does it accept values? startCrankingEngine(1) means startcranking and startCrankingEngine(nil) means no more cranking request?

rusefillc, 2024-10-15:
@ElDominio zero arguments

see https://github.com/rusefi/rusefi/wiki/Lua-Scripting#startcrankingengine

ElDominio, 2024-10-19:
[image]

Can't assign starter output if I don't choose a physical crank pin

ElDominio, 2024-10-19:
console says: "2024-10-19_16_23_28_537: EngineState: Let's crank this engine for up to 3 seconds via 6C - High Side 2!" 

but i get no starter? strange

rusefillc, 2024-10-19:
``EngineState: Let's crank this engine for up to 3 seconds via 6C - High Side 2!"`` is printed after toggling of physical output

@ElDominio please also share what console says few seconds later? wither way I would expect another line about disengaging starter?

rusefillc, 2024-10-19:
@ElDominio please use DMM or oscilloscope or on-board LED to confirm if physical cranking actually happens

ElDominio, 2024-10-19:
physical cranking is ok, it was me being a dumb. confirmed working

commit 7612d646 (2024-10-19): Lua 'startCrankingEngine' method fix #6983
closed 2024-10-19 by rusefillc