rusefi/rusefi #6991: uaefi121 ADC list incomplete

ggurov, 2024-10-21:
**Describe the bug**
There are very few labeled ADC inputs in TS, and lack of definition of remaining pins

**Hardware**
 - Board: uaefi121
 - Engine info ; bench

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'map sensor'
2. Click on 'map input'
3. Scroll down to lack of MAP sensor
4. See error of ways

**Expected behavior**
Ability to select ADC channels by their name

**Screenshots**
[image]

rusefillc, 2024-10-21:
not a solution, just a useful link for myself https://rusefi.com/docs/pinouts/hellen/uaefi121/

rusefillc, 2024-10-21:
random https://github.com/andreika-git/hellen-one/issues/394

rusefillc, 2024-10-21:
@ggurov just pushed info for five more pins, feel free to reopen if you can point on any more missed pins

closed 2024-10-21 by rusefillc