rusefi/rusefi #6999: Add variant of VW -  skoda favorit/felicia

ggurov, 2024-10-25:
**Is your feature request related to a problem? Please describe.**
It it somehow different from the existing 

**Would we need new configuration fields?**
New decoder that's a slight modification of the existing one for a variant of the engine

**Describe the solution you'd like**
Once tested, code incorporated for offset/etc

**Describe alternatives you've considered**
one off yolo firmware build

**Additional context**
WIP decoder attached.

ggurov, 2024-10-25:
[trigger_vw.cpp.txt](https://github.com/user-attachments/files/17528309/trigger_vw.cpp.txt)

rusefillc, 2024-10-26:
Not planning to add another trigger wheel

I agree that existing trigger seems to be not functional but adding a new one is not the way forward.

See also https://github.com/rusefi/rusefi/pull/6995

closed 2024-10-26 by rusefillc