rusefi/rusefi #7002: 3-0 dies for no reason at idle speeds

ElDominio, 2024-10-28:
**Describe the bug**
3-0 car with 6G72 cam is idling fine, then decides to quit. Logic logs shows no reason for this?

**Hardware**
 - Board: custom f407
 - Engine info 6g72

**To Reproduce**
Steps to reproduce the behavior:
1. turn negine on
2. idle for a while
3. watch triggerErrorCounter rise and engine die

**Expected behavior**
stay running

**Additional context**
ref: #6798

rusefillc, 2024-10-28:
MLV

[29scynerror.zip](https://github.com/user-attachments/files/17483548/29scynerror.zip)

at 29.6s ish,  see trigger error rise, but triggerPrimaryRise seems logical? who caused this error?

[image]

rusefillc, 2024-10-28:
logic data

[dyinglogiclogs.zip](https://github.com/user-attachments/files/17512243/dyinglogiclogs.zip)

I took a bunch of logs of the car having difficulty with cranking, and specifically, in "diedoutlog2.logicdata" at about 12.5s it has a sync loss event that leads to shutdown of the engine.

rusefillc, 2024-10-28:
[image]

rusefillc, 2024-10-28:
problem one: we seem to have (electrical?) noise

problem two: we've inherited unusually forgiving gap ratio settings from 12/0 trigger wheel while re-using 12/0 code

* recommending to fix root cause of electrical noise
* recommending to try custom less forgiving gap settings

length=2
from 0.7 to 1.3? that would hopefully handle noise differently enough?