rusefi/rusefi #7009: Basic updater shows incorrect branch if console is built in `GHA artifact mode`

kifir23917, 2024-10-29:
`Firmware at GHA` GHA has `LTS Build` parameter:

[image]

If this GHA is executed from `lts` branch, but with unchecked `LTS Build` checkbox, basic updater from resulting artifact displays `development` branch instead of `lts` branch:
 
[image]

Is it an issue?

kifir23917, 2024-10-29:
Example of such artifact can be found [here](https://github.com/rusefi/rusefi/actions/runs/11565087013).

kifir23917, 2024-10-29:
relates to #7000

rusefillc, 2024-10-29:
Real question seems to be "do we need the checkbox at all, or shall we remove it?"

At the moment code works as desired but it's a weird desire - if you use checkbox, it has priority