rusefi/rusefi #7011: AEM Wideband through CAN shows everything reading, but nothing on TS.

ElDominio, 2024-10-29:
**Describe the bug**
Everything indicated an AEM wideband is connected through CAN, but nothing shows on ts
**Hardware**
 - Board: custom f407
 - Engine info 22R

**To Reproduce**
Steps to reproduce the behavior:
1.  Connect wideband thru can
2. everything except gage works

**Expected behavior**
Wideband shows on ts

**Additional context**
https://rusefi.com/online/view.php?log=1432

ElDominio, 2024-10-29:
[image]

ElDominio, 2024-10-29:
[image]

ElDominio, 2024-10-29:
[image]

ElDominio, 2024-10-31:
00 th
2024-10-29_17_15_42_104: Engine State: VERBOSE Sending CAN bus0 message: ID=206/=8 7B 0 2F 0 0 0 00
2024-10-29_17_15_42_105: Engine State: VERBOSE Sending CAN bus0 message: ID=207/1-800000000
2024-10-29_17_15_42_108: EngineState: VERBOSE Sending CAN bus0 message: ID=208/1-8 0000 0000
2024-10-29_17_15_42_109: Engine State: VERBOSE Sending CAN bus0 message: ID=209/1-8 00000000
2024-10-29_17_15_42_111: Engine State: CANO RX: ID 0000180 DLC 8: 24 71 00 00 76 6A 02 04
2024-10-29_17_15_42_112: Engine State: VERBOSE Sending CAN bus0 message: ID=200/1-8 DB 1F 2F 23 16000
2024-10-29_17_15_42_114: Engine State: VERBOSE Sending CAN bus0 message: ID=201/1-8 1D 499 23 A 00
2024-10-29_17_15_42_116: Engine State: VERBOSE Sending CAN bus0 message: ID=202/1-8 9C FF 60 9C FF 9C FF
2024-10-29_17_15_42_118: Engine State: VERBOSE Sending CAN bus0 message: ID=203/1-8 14 4 70 15 28 28 3D 0
2024-10-29_17_15_42_121: Engine State: VERBOSE Sending CAN bus0 message: ID=204/1-8 0 0 0 0 28 28 24 35
2024-10-29_17_15_42_123: Engine State: VERBOSE Sending CAN bus0 message: ID=205/1-8 640 F2 4 C3 1 00
2024-10-29_17_15_42_124: Engine State: VERBOSE Sending CAN bus0 message: ID=206/=8 7B 0 2F 0 0 0 00
2024-10-29_17_15_42_126: Engine State: VERBOSE Sending CAN bus0 message: ID=207/1=800000000
2024-10-29_17_15_42_127: Engine State: VERBOSE Sending CAN bus0 message: ID=208/1=800000000
2024-10-29_17_15_42_129: Engine State: VERBOSE Sending CAN bus0 message: ID=209/1=800000000
2024-10-29_17_15_42_131: Engine State: VERBOSE Sending CAN bus0 message: ID=200/1-8 DB 1F 2F 23 16000
2024-10-29_17_15_42_133: Engine State: VERBOSE Sending CAN bus0 message: ID=201/1-8 1D 49C 23 A 00
2024-10-29_17_15_42_134: Engine State: VERBOSE Sending CAN bus0 message: ID=202/1-8 9C FF B 0 9C FF 9C FF
2024-10-29_17_15_42_135: Engine State: VERBOSE Sending CAN bus0 message: ID=203/1-8 14 4 70 15 28 28 3D 0
2024-10-29_17_15_42_138: Engine State: VERBOSE Sending CAN bus0 message: ID=204/1-8 0 0 0 0 28 28 FF 34
2024-10-29_17_15_42_139: Engine State: VERBOSE Sending CAN bus0 message: ID=205/1-8 640 F2 4 C3 1 00
2024-10-29_17_15_42_141: Engine State: VERBOSE Sending CAN bus0 message: ID=206/1-8 7B 0 2F 000 00
2024-10-29_17_15_42_143: Engine State: VERBOSE Sending CAN bus0 message: ID=207/1=800000000
2024-10-29_17_15_42_144: Engine State: VERBOSE Sending CAN bus0 message: ID=208/1=800000000
2024-10-29_17_15_42_146: Engine State: VERBOSE Sending CAN bus0 message: ID=209/1-8 00000000
2024-10-29_17_15_42_147: Engine State: CANO RX: ID 0000180 DLC 8: 24 1F 00 00 79 6E 02 04
2024-10-29_17_15_42_149: Engine State: WBO SERVICE Sending CAN bus0 message: ID=EF50000/-2 87 1000000
2024-10-29_17_15_42_151: EngineState: VERBOSE Sending CAN bus0 message: ID=200/1-8 DB 1F 2F 23 16000
2024-10-29_17_15_42_154: Engine State: VERBOSE Sending CAN bus0 message: ID=201/1-8 1D 4 A4 13 A 00
2024-10-29_17_15_42_156: Engine State: VERBOSE Sending CAN bus0 message: ID=202/1-8 9C FF B 0 9C FF 9C FF
2024-10-29_17_15_42_157: EngineState: VERBOSE Sending CAN bus0 message: ID=203/1-8 D 4 70 15 28 28 3D 0
2024-10-29_17_15_42_159: Engine State: VERBOSE Sending CAN bus0 message: ID=204/-8 0 0 0 0 28 28 F4 34

rusefillc, 2024-11-03:
@FDSoftware next step would be a unit test to see how "ID 0000180 DLC 8: 24 1F 00 00 79 6E 02 04" is decoded by AEM CAN consumer

which would probably (have not looked at the code yet) depend on refactoring that class to make it unit-testable?

ElDominio, 2024-11-03:
english pls

FDSoftware, 2024-11-03:
testing the frame: `ID 0000180 DLC 8: 24 71 00 00 79 6E 02 04` i got:

[image]
and:

`ID 0000180 DLC 8: 24 1F 00 00 79 6E 02 04` 
[image]

The code behaves as expected in the AEM manual, the probe is reporting that the lambda values ​​are not valid:
[image]

rusefillc, 2024-11-03:
@FDSoftware progress! Is the code ready to cover this frame with a unit test? Do we already have a unit test for aem decoding?

rusefillc, 2024-11-03:
@FDSoftware do we have a unit test for invalid AEM decode? I do not see one :(

rusefillc, 2024-11-03:
@ElDominio see adjusted item summary

I've just pushed potential usability improvement but overall root cause AEM says stuff is broken on AEM side?

ElDominio, 2024-11-26:
> testing the frame: `ID 0000180 DLC 8: 24 71 00 00 79 6E 02 04` i got: [...]

04 in the last byte doesn't mean anything though, 02 would be bad sensor?

ElDominio, 2024-11-26:
Why can't we just report errors, and still use reported Lambda values?

rusefillc, 2024-11-26:
@ElDominio you have Lua for that?

Looks like we do not know enough about _why_ error exist to decide if we can just hide it

FDSoftware, 2024-11-29:
> 04 in the last byte doesn't mean anything though, 02 would be bad sensor?

We don't know, it falls within the reserved bit of AEM, 02 represents an invalid lambda value (which could happen in the free-air calibration, or for reasons that AEM does not report [this information could be within 04 ?])

ElDominio, 2024-11-29:
> @ElDominio you have Lua for that?
> 
> Looks like we do not know enough about _why_ error exist to decide if we can just hide it

I think that unless errors our reported by a rusefi-based device, we should still display lambda. By choosing not to display the data that is shown to the user by the gauge itself, OUR stuff is being flagged as the non-working component. 
"But my gauge shows numbers! Why does your shitty ECU not show numbers Norman!?!?!? I will use Haltech that shows me the numbers"

It is not for us to decide if the number is correct, the user should have an indication that the gauge says the number is not trustworthy (and maybe disable closed loop based on that?), but disabling the number in logs only directs the problem to us instead of their hardware.

TL;DR AFR gauge value should be shown and so should the error. Doing otherwise causes severe useability issues

rusefillc, 2025-06-29:
@ElDominio i know we had major clean-up arounf AEM vs RE WBO, what's the state of this ticket?

ElDominio, 2025-06-29:
I am pretty sure this is working quite well. @ankyth could ocmment on this but the choice of being AEM CAN vs RusEFI CAN has impvoed usablity. I think this can be closed

closed 2025-06-30 by rusefillc